Bodmer/TFT_eSPI #1132: initDMA conflicts with non-DMA operations

shb3014, 2021-04-13:
Once `initDMA` is called with `ctrl_cs` argument as true, all non-DMA operations afterwards seem to be incorrect.
I'm using a 240*320 st7789v screen with esp32 (platfomio with arduino framework, esp32-wrover-e with psram enabled).
`TFT_RGB_ORDER` is defined as `TFT_BGR`, `TFT_RST` is set to -1, and `SPI_FREQUENCY` is defined as 80000000.
The following is a code snippet to replicate the problem.
    
    Serial.begin(115200);
    tft.init();
    tft.fillScreen(TFT_BLACK);
    tft.initDMA(true);
    delay(1000);
    tft.fillScreen(TFT_WHITE);

Several blue lines appear on white filled background.
[image]

Setting `ctrl_cs` as false would fix the problem but then dma is not useable anymore.

One way I found to make both work is to set `ctrl_cs` as false, and put `CS_L` and `CS_H` in `pre_cb` and `post_cb` when initializing the spi bus config, so that cs is controlled at a different layer. 

    void cs_high(spi_transaction_t *){
        CS_H;
    }    
    void cs_low(spi_transaction_t *){
        CS_L;
    }

The following goes in initDMA in the processor file.

    spi_device_interface_config_t devcfg = {
      .command_bits = 0,
      .address_bits = 0,
      .dummy_bits = 0,
      .mode = TFT_SPI_MODE,
      .duty_cycle_pos = 0,
      .cs_ena_pretrans = 0,
      .cs_ena_posttrans = 0,
      .clock_speed_hz = SPI_FREQUENCY,
      .input_delay_ns = 0,
      .spics_io_num = pin,
      .flags = SPI_DEVICE_NO_DUMMY, //0,
      .queue_size = 1,
      .pre_cb = cs_low, //dc_callback, //Callback to handle D/C line
      .post_cb = cs_high
    };
However, I'm still not sure about the cause of this conflict.

Bodmer, 2021-04-13:
Which version are you having a problem with?

shb3014, 2021-04-14:
@Bodmer I'm using the latest version on main branch.

Bodmer, 2021-04-16:
I am unable to reproduce the problem with the Arduino IDE ( I do not use PlatformIO, but it should work the same).

My test setup is: Setup23_TTGO_TM.h running SpriteRotatingCube unaltered is working OK, then changed[ line 166](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/DMA%20test/SpriteRotatingCube/SpriteRotatingCube.ino#L166) to:
```
  tft.initDMA(5);
```
Still runs fine, no problems see at all.

I suspect a black screen may be the power up default, so it could be the the TFT_CS line is not defined? Run "Read_User_Setup" diagnostic sketch and check the serial output is correct. Post the output here so I can see exactly what your setup configuration is. Also what version of the ESP32 board package are you using? (I am using 1.0.6)

Bodmer, 2021-04-16:
I just realised SpriteRotatingCube only uses DMA and no graphics commands so was not a good test! However I repeated the test with Flash_Jpg_DMA and that runs fine as well with tft.initDMA(true);

shb3014, 2021-04-19:
This is the output for my setup.
```
TFT_eSPI ver = 2.3.64
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 240
Display height = 320

MOSI    = GPIO 19
SCK     = GPIO 21
TFT_CS   = GPIO 23
TFT_DC   = GPIO 22

TFT_BL           = GPIO 18
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 80.00
```

I'm not sure where to look for the ESP32 board package though.

In order to replicate the issue again, I simply changed the setup of  "Read_User_Setup" to the following.
```
void setup() {
    // Use serial port
    Serial.begin(115200);

    // Initialise the TFT screen
    tft.init();
    /* before tft initDMA, fill should work without problem */
    tft.fillScreen(TFT_WHITE);
    tft.initDMA(true);
    delay(1000);
    /* after the tft initDMA(true), fill gives blue lines to the screen */
    tft.fillScreen(TFT_WHITE);
}
```
On my esp32 with the above setup, the issue persists. 
Thanks!

Bodmer, 2021-04-19:
I suspect that the reason for the problem is the choice of SPI pins and the high SPI frequency.

The ESP32 datasheet contains warnings about using SPI to non-native pins since then they are routed through a multiplexer which causes timing issues. This is probably exacerbated by the DMA engine setup.

Try a lower SPI clock frequency and ideally the default native SPI pins.

If you use ```tft.initDMA();``` then DMA can be using ```tft.startWrite();```  which locks CS low and ```tft.endWrite();``` to release CS after the DMA is complete. See[ example here](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/DMA%20test/boing_ball/boing_ball.ino#L132-L173). Does that example run OK with ```tft.initDMA(true);``` set?

These are the pins I am using:
```
#define TFT_MOSI 23
#define TFT_SCLK 18

#define TFT_CS   05
#define TFT_DC   16
#define TFT_RST  17
```

shb3014, 2021-04-21:
I tested with a lower frequency and it worked flawlessly. However I can't easily reroute the IOs to verify how much a role it plays in this. 
What I still don't get is that why adding CS_L and CS_H in pre_cb and post_cb  in the processor file will make both work in high frequency. I guess using them in callbacks just slows the spi down comparing to set spics_io_num in the config.
Really appreciate your time. It seems the issue has been resolved. This is a great project and enabled me to make things I thought unimaginable.  Thanks again for the efforts. Please feel free to close this issue.

Bodmer, 2021-04-21:
Glad to hear it is working.

closed 2021-04-21 by Bodmer