Bodmer/TFT_eSPI #1147: Update to 2.3.64 broke my sprite positioning

LiveRock, 2021-04-20:
```
spr.loadFont(DigitalLCDFont); 
  tft.setTextDatum(TC_DATUM);
  // Channel 1
  spr.setTextColor(TFT_DARKCYAN, meterBGColour); // Set the sprite font colour and the background colour
  tft.setCursor(40, 70);          // Set the tft cursor position, yes tft position!
  spr.printToSprite(String(displayChannel1));    // Prints to tft cursor position, tft cursor NOT moved
  
  // Channel 2
  spr.setTextColor(TFT_DARKGREY, meterBGColour); // Set the sprite font colour and the background colour
  tft.setCursor(40, 150);          // Set the tft cursor position, yes tft position!
  spr.printToSprite(String(displayChannel2)); 

  spr.unloadFont();
```

It was working before but after updating to the latest, the spr.printToSprite command show the sprite at the top left hand corner. Nothing I can do can make it back.
Any suggestions?

LiveRock, 2021-04-20:
tft.setCursor does not do anything. What am i doing wrong here?

LiveRock, 2021-04-21:
Issue resolved. Due to not enough memory

closed 2021-04-21 by LiveRock

Bodmer, 2021-04-22:
Thanks for letting me know it is resolved.