Bodmer/TFT_eSPI #1152: Display stops working in 8-bit parallel mode

Jakub-Nagy, 2021-04-24:
Hi! I'm building a smart device with a touch screen with a few other people. We've encountered this weird issue, we hope someone here will be able to help us.

**Hardware**
We're using a ESP32-PICO-D4,

 along with a 320x480px LCD panel with the ILI9488 driver and a CTP panel. The display is connected using 8-pin parallel interface.

**Platform**
We are using the PlatformIO development environment, along with the Arduino software framework, along with the [Espressif 32 platform](https://github.com/platformio/platform-espressif32/releases) for PlatformIO.

**Libraries**
 - [TFT eSPI](https://github.com/Bodmer/TFT_eSPI) as the display driver.
 - [Arduino GT911](https://github.com/u4mzu4/Arduino_GT911_Library) as the touch driver.
 - [LVGL](https://github.com/lvgl/lvgl) as the interface library.

**Tested Firmware**
We have tested 2 different pieces of firmware:

- **display testing firmware** is very simple, it only draws a white square on the current detected touch point.
- **final interface firmware** is much more complex, it invokes LVGL, uses Wi-Fi and so on.

**Problem Description**
The original version of the final interface firmware was developed over a year ago. There were some problems with the RTP panel and the SPI connection to the display driver, so we changed the hardware, but this has taken a long time. I received the corrected hardware a couple of weeks ago. I created the display testing firmware for the purpose of testing the new hardware. It worked well, so I passed the device to the programmer who is developing the final interface firmware.

He was able to successfully compile, flash and run the display testing firmware. When he ran the final interface firmware, the display was always clear and white. The display config was correct. The library version was the same. After some time, he updated the Espressif 32 platform, because it was outdated. After this, neither of the testing firmwares worked, even after specifying older platform version.

I know that it might be hard to find the issue just from my description, let me know if you need more information. Is there a debug mode in the library, which we could use to possibly identify the issue? Or do we need an oscilloscope to test the interface? Thanks in advance.

Bodmer, 2021-04-24:
Since the hardware was working OK with test code then the problem is most likely somewhere in the software and/or software environment.

Note that if you update TFT_eSPI that the setup files will be over-written, however in PlatofrmIO typically users incorporate the settings in their project "platformio.ini" file. [Example here](https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/PlatformIO/Configuring%20options.txt).

In these "it was working, now it is not" situations it is best to backtrack and go back to a simpler setup and keep this software/environment setup archived so that it can be reinstated to verify the hardware still works with library examples.

I have only used lvgl for some quick tests but note that the API has been updated over the past 12 months.

Jakub-Nagy, 2021-04-27:
> Since the hardware was working OK with test code then the problem is most likely somewhere in the software and/or software environment.

Okay, we ditched most of the things that could cause problems. We used no LVGL, touch control or PlatformIO. We just used this library along with our configuration file and the default examples. We tried game of life and rainbow, neither of them worked.

Please check the configuration file, is there any clear problem with it when running a paralel setup?
[User_Setup.txt](https://github.com/Bodmer/TFT_eSPI/files/6382391/User_Setup.txt)

RudyFiero, 2021-04-29:
Game of life didn't work for me (I forget why) so I just moved on. I also am using 8 bit parallel mode with an ESP32. I am using a XPT2046 resistive touch controller. I am able to run LVGL version 7 widget examples and the following code. https://www.youtube.com/watch?v=AvJms6_Pf_o (though the Ping code causes a crash)
https://github.com/0015/ThatProject/blob/master/ESP32_LVGL/DynamicWIFIConnector_LVGL7/DynamicWIFIConnector_LVGL7.ino

Bodmer, 2021-05-03:
The setup file looks OK. For reference [here is the setup file I used for testing](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup16_ILI9488_Parallel.h).

It sounds like even the "display testing firmware" which used to work does not anymore? In that case it is difficult to advise what has gone wrong. Maybe a broken  software environoment or damaged/faulty hardware.

Unfortunately it is not possible to debug these situations without hardware access, so I will have to close.

I hope you manage to find where the problem is. I would focus on getting that display testing firmware running again.

closed 2021-05-03 by Bodmer