Bodmer/TFT_eSPI #1159: Text and poligons in the same line

nicoworq, 2021-04-27:
Hi! First of all, thanks a lot for this magnificent library! 

I want to display some text followed by a triangle, all in the same line. 

For example : 

SampleString  ▲  | SampleString2  ▼  

I was trying to do this with drawString(), but the length of the string may vary, so i don't know how to get the correct cursor position to draw the triangle after the text. 

Is there any way to do this or something similar? 

Thanks a lot! 

Have a great day!

Bodmer, 2021-04-27:
drawString returns the pixel width of the string, so:
```
int16_ t w = drawString("Hello", x, y);
```
returns the pixel width of "Hello" in w, so then the end is at coordinate x+w, y

nicoworq, 2021-04-28:
Thanks a lot!

closed 2021-04-28 by nicoworq