Bodmer/TFT_eSPI #1163: ESP32S + ISI9341 no touch

sirEgghead, 2021-04-29:
I'm unable to get the touch to work on my 9341 SPI and an ESP32S using the library.  When I run the touch calibration, I don't get a chance to touch the corners before the sketch runs through.

I connected only the pins listed in the config below + 3v3 and gnd.

```
ets Jun  8 2016 00:22:57

rst:0x9 (RTCWDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

[code]
TFT_eSPI ver = 2.3.66
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 21

TFT_BL           = GPIO 32
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50
[/code]

```

Bodmer, 2021-04-29:
Have you wired up the touch controllers pins to the SPI bus and touch chip select?

Bodmer, 2021-04-29:
Pin names vary, these are the typical connections:

T_DO -> MISO pin (e.g. TFT_MISO)
T_DIN -> MOSI pin (e.g. TFT_MOSI)
T_CS -> pin of you choice specified in setup file as TOUCH_CS
T_CLK -> SCLK pin (e,g, TFT_SCLK)
T_IRQ -> not connected

sirEgghead, 2021-04-29:
That was it.  I figured they needed to be connected somewhere, but I didn't realize they were piggybacked off the other pins.  Makes sense though since it is serial.

I appreciate the help.  I'm learning by trial.  I'm used to working with parallel, but I wanted to give SPI a try for various reasons.  Great library!

closed 2021-04-29 by sirEgghead