Bodmer/TFT_eSPI #1171: GC9A01  tft.readRect() problem

serdargoksal, 2021-05-05:
Hi Bodmer,

While using GC9A01 display driver.
tft.readRect() does not  work  when the line  
             #define USE_HSPI_PORT
   is commented in User_Setup.h.

When that line is uncommented, the above function works properly but tft.drawLine() and tft.drawText()   functions are not working.

so we are unable to run your  animated_dial.ino example as expected ;)

Please advise,

we are using Bodmers example TFT_Read_Write.ino

pin configurations are;
#define TFT_MISO 19    // In some display driver board, it might be written as "SDA" and so on.
#define TFT_MOSI 23    // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 18
#define TFT_CS   5       // Chip select control pin
#define TFT_DC   26    // Data Command control pin
#define TFT_RST  27    // Reset pin (could connect to Arduino RESET pin)

#define TFT_BL   22   // LED back-light

Bodmer, 2021-05-06:
The display board  I have does not provide the MISO pin and does not operate in  MOSI bidirectional mode and thus cannot be read. Can you provide a web link to the display you have.

What SPI clock speed are you using.

serdargoksal, 2021-05-06:
Here is the link of the display
https://www.aliexpress.com/item/1005001382306772.html?spm=a2g0s.9042311.0.0.27424c4d4qY5MR

You are correct as a mistake I defined MISO pin.

it is very strange that when   #define USE_HSPI_PORT defined. your code reads the mentioned area from the screen.
and I can push.image() anywhere in the secreen without a problem.

all the best

Bodmer, 2021-05-06:
Diverting the HSPI port to the VSPI default pins introduces timing delays so I suspect this is sufficient to make the read work. However the library expects the VSPI port to be used for bidirectional transfer so normal writes are probably then getting messed up. The library uses the trick of moving the MISO pin to the MOSI pin during the rreD phase so this is probably the reason.

I will investigate further. That display has other issues too at the moment and so will not be declared as supported formally  supported until those have been investigated.

closed 2021-11-28 by Bodmer