Bodmer/TFT_eSPI #1172: ILI9341_Init.h - Strange graphic issues depending on TFT

Katana1234, 2021-05-06:
Hi,

i just came across some strange issues...

I have 3 different ILI9341 Displays, to make it easier i will call them by the color:
RED: https://www.amazon.de/gp/product/B017FZTIO6/
-> This is the most common ILI9341 i guess

BLACK: https://wiki.dfrobot.com/2.8inches_320_240_IPS_TFT_LCD_Touchscreen_SKU_DFR0665
-> DFROBOT's version with an FPC-connector + built in BL-dimming-pin

BLUE: https://www.amazon.de/gp/product/B07QJWXPJT
-> Another version with built in BL-dimming-pin

When i first tested the black one i though the display was even cheaper than the other ones, because it gave me some strange lines when looking to the tft from the left side (assuming that the pin header belongs to the left).
Its really hard to take propper pictures of the problem because of the view angle + anti aliasing and other photo related stuff. ;-)
But this is what you have to expect:
[image]

There was also another problem that looked like a lcd-burin or lcd-memory-effect problem. The display would burin a screen after just a couple of minutes. In this picture a screen was showing for 10 minutes, then i switched to solid yellow to make the "burnin" visible:
[image]

But before throwing it into the trash i wanted to make sure it is a hardware problem. So i used DFROBOT's library instead if this one. And i was suprised to see that the strange lines disappeared.
[image]

So i compared even the other displays to make sure they don't show a similar behaviour.
And the sad thing ist: Even the other displays show those strange lines when looking from a steep angle. The only difference is that you really have to look closely to recognize them.

Then i compared the difference in the inits of the 2 libraries:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9341_Init.h
https://github.com/DFRobot/DFRobot_GDL/blob/master/src/Drivers/DFRobot_ILI9341.h
And there is one line that fixes this issue in the DFR lib, the VCM control 2, also called VCOM control 2.
DFR lib sends: 0xB7
TFT_eSPI sends: 0x86

So, when replacing ILI9341_Init.h line 55 with:
` writedata(0xB7); `
The issue is gone. And the issue that looked like a lcd-burnin was also gone. :)

Replacing only this line fixed those issues, but i still had to mix both init files to get back a propper gamma setting.

Everything compared:
Black screen:
[image]
[image]
[image]

Red screen:
[image]
[image]

Blue screen:
[image]
[image]

Am i the only one having these issues? :)

Bodmer, 2021-05-06:
Thanks for reporting and investigating this problem. There is some variation in the required init values between the TFT panel manufacturers even though the same controller is used. The further complication is that panels may be fitted with clone chip chips when bought from some suppliers in China. Yet another complication is production process variations and improvements ver the years that change the electrical characteristics slightly. The only way around this is to buy from screen suppliers that guarantee genuine parts and provide the optimised init code for the actual display sold.

The current init sequence is the same as used by Adafruit ILI9341 library which was the starting seed for this library.

In the ILI9481 case the library provides 6 different init sequences, a generic one plus 5 other options!

I will add a new option for the ILI9341.

Katana1234, 2021-05-06:
Thanks for this clarification @Bodmer  :)

If possible, please use this config as a reference. It is a mix of the original one and the DFR one combining several settings like the gamma settings.
Working with all 3 types of my display!

`
{
  writecommand(0xCF);
  writedata(0x00);
  writedata(0XC1);
  writedata(0X30);

  writecommand(0xED);
  writedata(0x64);
  writedata(0x03);
  writedata(0X12);
  writedata(0X81);

  writecommand(0xE8);
  writedata(0x85);
  writedata(0x00);
  writedata(0x78);

  writecommand(0xCB);
  writedata(0x39);
  writedata(0x2C);
  writedata(0x00);
  writedata(0x34);
  writedata(0x02);

  writecommand(0xF7);
  writedata(0x20);

  writecommand(0xEA);
  writedata(0x00);
  writedata(0x00);

  writecommand(ILI9341_PWCTR1); //Power control
  writedata(0x10);              //VRH[5:0]

  writecommand(ILI9341_PWCTR2); //Power control
  writedata(0x00);              //SAP[2:0];BT[3:0]

  writecommand(ILI9341_VMCTR1); //VCM control
  writedata(0x30);
  writedata(0x30);

  writecommand(ILI9341_VMCTR2); //VCM control2
  writedata(0xB7);              //--

  writecommand(ILI9341_PIXFMT);
  writedata(0x55);

  writecommand(0x36); // Memory Access Control
  writedata(0x08);    // Rotation 0 (portrait mode)

  writecommand(ILI9341_FRMCTR1);
  writedata(0x00);
  writedata(0x1A);

  writecommand(ILI9341_DFUNCTR); // Display Function Control
  writedata(0x08);
  writedata(0x82);
  writedata(0x27);

  writecommand(0xF2); // 3Gamma Function Disable
  writedata(0x00);

  writecommand(0x26); //Gamma curve selected
  writedata(0x01);

  writecommand(0xE0); //Set Gamma
  writedata(0x0F);
  writedata(0x2A);
  writedata(0x28);
  writedata(0x08);
  writedata(0x0E);
  writedata(0x08);
  writedata(0x54);
  writedata(0xA9);
  writedata(0x43);
  writedata(0x0A);
  writedata(0x0F);
  writedata(0x00);
  writedata(0x00);
  writedata(0x00);
  writedata(0x00);

  writecommand(0XE1); //Set Gamma
  writedata(0x00);
  writedata(0x15);
  writedata(0x17);
  writedata(0x07);
  writedata(0x11);
  writedata(0x06);
  writedata(0x2B);
  writedata(0x56);
  writedata(0x3C);
  writedata(0x05);
  writedata(0x10);
  writedata(0x0F);
  writedata(0x3F);
  writedata(0x3F);
  writedata(0x0F);

  writecommand(0x2B);
  writedata(0x00);
  writedata(0x00);
  writedata(0x01);
  writedata(0x3f);

  writecommand(0x2A);
  writedata(0x00);
  writedata(0x00);
  writedata(0x00);
  writedata(0xef);

  writecommand(ILI9341_SLPOUT); //Exit Sleep

  end_tft_write();
  delay(120);
  begin_tft_write();

  writecommand(ILI9341_DISPON); //Display on
}
`

Onefabis, 2021-05-08:
Hi, perhaps in my case my screen ILI9225 (8 bit parallel model) has the same nature of the problem but I don't know for sure 
[image]

Onefabis, 2021-05-08:
Hah, here is my small victory. I modified ILI9225_Init.h by myself and find best combitations of OSC, frame frequency and Power modes, so here are line you need to tweak for good display balance. But pay attention: it's good only for display with my parameters: 2inch 8 bit parallel ILI9225. 
[image]
So change those lines in ILI9225_Init.h:

18 line replace from:
writedata(0x00);writedata(0x18);
to:
writedata(0x00);writedata(0x1F);

20 line replace from:
writedata(0x61);writedata(0x21);
to:
writedata(0x62);writedata(0x86);

26 line replace from:
writedata(0x08);writedata(0x00);
to:
writedata(0x0F);writedata(0x00);

42 line replace from:
writedata(0x11);writedata(0x00);
to:
writedata(0x11);writedata(0x07);

46 line replace from:
writedata(0x0D);writedata(0x01);
to:
writedata(0x0F);writedata(0x01);

I make a test with ESP32 WROOM 32 (38 pins):

TFT_eSPI ver = 2.3.61
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = Parallel
Display driver = 9225
Display width  = 176
Display height = 220

TFT_CS   = GPIO 33
TFT_DC   = GPIO 15
TFT_RST  = GPIO 32
TFT_WR   = GPIO 4
TFT_RD   = GPIO 2

TFT_D0   = GPIO 23
TFT_D1   = GPIO 5
TFT_D2   = GPIO 18
TFT_D3   = GPIO 19
TFT_D4   = GPIO 17
TFT_D5   = GPIO 16
TFT_D6   = GPIO 22
TFT_D7   = GPIO 21

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Bodmer, 2021-05-11:
@Onefabis please raise a new issue, this one is for ILI9341.

closed 2021-05-11 by Bodmer

Katana1234, 2021-05-17:
> I have added an option to use #define ILI9341_2_DRIVER in the setup. please test.

working as expected. Thank you very much!!! :)

Bodmer, 2021-05-17:
Thanks for letting me know.

brightproject, 2023-06-04:
> Hi,
> 
> i just came across some strange issues... Everything tested with ESP32.
> 
> I have 3 different ILI9341 Displays, to make it easier i will call them by the color: RED: https://www.amazon.de/gp/product/B017FZTIO6/ -> This is the most common ILI9341 i guess
> 
> BLACK: https://wiki.dfrobot.com/2.8inches_320_240_IPS_TFT_LCD_Touchscreen_SKU_DFR0665 -> DFROBOT's version with an FPC-connector + built in BL-dimming-pin
> 
> BLUE: https://www.amazon.de/gp/product/B07QJWXPJT -> Another version with built in BL-dimming-pin
> 
> When i first tested the black one i though the display was even cheaper than the other ones, because it gave me some strange lines when looking to the tft from the left side (assuming that the pin header belongs to the left). Its really hard to take propper pictures of the problem because of the view angle + anti aliasing and other photo related stuff. ;-) But this is what you have to expect: [image]
> 
> There was also another problem that looked like a lcd-burin or lcd-memory-effect problem. The display would burin a screen after just a couple of minutes. In this picture a static screen was showing for 10 minutes, then i switched to solid yellow to make the "burnin" visible: [image]
> 
> But before throwing it into the trash i wanted to make sure it is a hardware problem. So i used DFROBOT's library instead of this one. And i was suprised to see that the strange lines disappeared. [image]
> 
> So i compared even the other displays to make sure they don't show a similar behaviour. And the sad thing is: Even the other displays show those strange lines when looking from a steep angle. The only difference is that you really have to look closely to recognize them.
> 
> Then i compared the difference in the inits of the 2 libraries: https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9341_Init.h https://github.com/DFRobot/DFRobot_GDL/blob/master/src/Drivers/DFRobot_ILI9341.h And there is one line that fixes this issue in the DFR lib, the VCM control 2, also called VCOM control 2. DFR lib sends: 0xB7 TFT_eSPI sends: 0x86
> 
> So, when replacing ILI9341_Init.h line 55 with: `writedata(0xB7);` The issue is gone. And the issue that looked like a lcd-burnin was also gone. :)
> 
> Replacing only this line fixed those issues, but i still had to mix both init files to get back a propper gamma setting.
> 
> Everything compared: Black display showing stripes: [image] Black display showing burnin: [image] Black display fixed: [image]
> 
> Red display showing stripes: [image] Red display fixed: [image]
> 
> Blue display showing stripes: [image] Blue display fixed: [image]
> 
> Am i the only one having these issues? :) And what is the init sequence for the ILI9341 based on?

Hello, I have a problem initializing the display with the driver *
There are commands from the manufacturer, but none of them start the display.
I decided to compare the datasheets for ILI9341 and ILI9342C
I was surprised that ILI9342 does not have a physical contact EXTC
https://github.com/Bodmer/TFT_eSPI/issues/2682