Bodmer/TFT_eSPI #1173: dynamic selection of the chip

infrafast, 2021-05-06:
Hello, 
I face the situation to have 2 possible ST7735 chip: with or without "S" at the end of the reference depending on supplier.

I understand I have to change the setting file to use the one best fitting the need, and it works
 //#define ST7735_INITB
 //#define ST7735_GREENTAB
#define ST7735_GREENTAB2				//for ST7735
 //#define ST7735_GREENTAB3
// #define ST7735_GREENTAB128    // For 128 x 128 display
// #define ST7735_GREENTAB160x80 // For 160 x 80 display (BGR, inverted, 26 offset)
// #define ST7735_REDTAB				//for ST 7735S
 //#define ST7735_BLACKTAB
 //#define ST7735_REDTAB160x80   // For 160 x 80 display with 24 pixel offset

This has however to be done static when compiling the code.

2 questions / requirement proposal:
1) is there a way to query the chip and see which one equiped the board, so I would have something like getChip() 
2) and/or Is there a way or a chance to do the chip selection dynamically during run time using a function such setChip(...) to abvoid having comoiling the code for each of them ?

The use case is that there are many device in the field, some of them are with the S , some of them are not ...

Thanks for your thought and possible fix that may go in that direction

Bodmer, 2021-05-06:
Yes, for ST7735 only this is possible. See here:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L391

closed 2021-05-06 by Bodmer

infrafast, 2021-05-06:
thank you! 
I've commented all line related to the model in User_setup.h 
 //#define ST7735_INITB
 //#define ST7735_GREENTAB
//#define ST7735_GREENTAB2				//for ST7735
 //#define ST7735_GREENTAB3
// #define ST7735_GREENTAB128    // For 128 x 128 display
// #define ST7735_GREENTAB160x80 // For 160 x 80 display (BGR, inverted, 26 offset)
// #define ST7735_REDTAB				//for ST 7735S
 //#define ST7735_BLACKTAB
 //#define ST7735_REDTAB160x80   // For 160 x 80 display with 24 pixel offset

then change my call to init() to  tft.init(INITR_GREENTAB2);    // I used to use ST7735_GREENTAB2 for ST7735 and ST7735_REDTAB 
 for ST7735S

maybe I miss something but it has no effect, it looks like it keep using the same code no matter I call init with one colorcode or another... did I miss a #define or something ?
Thanks for your feedback

infrafast, 2021-05-10:
sorry to bother, any chance to review the above mentionned issue?

Bodmer, 2021-05-10:
It works OK for me. Are you using the Arduino IDE?

You still need to declare the display driver type in your setup file, so do not comment that out:
```
#define ST7735_DRIVER
```
The different tab colour [init sequences are selected here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ST7735_Init.h#L143-L198) by the tabcolor defined in the init(), so you can add debug print lines to test if the correct sequences are being invloked. e.g. like this:
```
...
       if (tabcolor == INITR_GREENTAB)
       {
         Serial.print("INITR_GREENTAB");
         commandList(Rcmd2green);
         colstart = 2;
         rowstart = 1;
       }
       else if (tabcolor == INITR_GREENTAB2)
       {
         Serial.print("INITR_GREENTAB2");
         commandList(Rcmd2green);
         writecommand(ST7735_MADCTL);
         writedata(0xC0 | TFT_MAD_COLOR_ORDER);
         colstart = 2;
         rowstart = 1;
       }
...
```
Since I cannot reproduce it I think you need to try to debug the problem.