Bodmer/TFT_eSPI #1176: Mixed colors on ST7735

WojakGra, 2021-05-06:
As i decribed colors are mixed at screen. Black is showing as black but orange is showing as blue.

Bodmer, 2021-05-06:
Typically this happens if the wrong "tab" colour option has been chosen. Originally the ST7735 variants were identified by the colour of the tab on the screen protection film. This is no longer the case so you need to try the other options.

WojakGra, 2021-05-06:
What do you mean by other options ?

Bodmer, 2021-05-06:
These options are in the setup file here:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L66

And here:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L86

WojakGra, 2021-05-06:
Ok, thank you. The "#define TFT_RGB_ORDER TFT_BGR" worked for me :)

closed 2021-05-06 by WojakGra