Bodmer/TFT_eSPI #1182: Problem compiling code for esp32-s2

DMohammed, 2021-05-09:
I’m starting to switch from esp32 to esp32-s2. When compiling I get an error like this: "VSPI was not declared in this scope." Does anyone know what that might be about and how to fix it? Thanks in advance.

Bodmer, 2021-05-09:
See #807

DMohammed, 2021-05-10:
I'm change VSPI to HSPI and set pin in this order 
```
#define TFT_MISO 37
#define TFT_MOSI 35
#define TFT_SCLK 36
#define TFT_CS   34  // Chip select control pin
#define TFT_DC    38  // Data Command control pin
#define TFT_RST   39 // Reset pin (could connect to RST pin)
```
example code compile but not working
my lcd model is ili9341 and set to spi mode

Bodmer, 2021-05-10:
Unfortunately I do not have an ESP32-S2 so am unable to advise further. At the moment the library makes no attempt to support that processor variant. Try a post in #807 and see if you can get a response from @nikthefix.

closed 2021-05-10 by Bodmer

erichiller, 2021-06-07:
I have this working now. Add this in your `User_Setup.h` -- no further changes are required. 
```c
#define VSPI SPI2_HOST
```

My setup is ILI9481_DRIVER and Adafruit ESP32-S2 Metro.

`User_Setup.h`:

```c

// NOTE:

// Parallel bus is only supported for the STM32 and ESP32
// Example below is for ESP32 Parallel interface with UNO displays
// #define USE_HSPI_PORT
#define TFT_PARALLEL_8_BIT
#define VSPI SPI2_HOST

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
#define ILI9481_DRIVER

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS A3     // Chip select control pin (library pulls permanently low
#define TFT_DC A2     // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST A4    // Reset pin, toggles on startup

#define TFT_WR A1    // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD A0    // Read strobe control pin

#define TFT_D0 13    // Must use pins in the range 0-31 for the data bus
#define TFT_D1 14    // so a single register write sets/clears all bits.
#define TFT_D2 7     // Pins can be randomly assigned, this does not affect
#define TFT_D3 8     // TFT screen update performance.
#define TFT_D4 9
#define TFT_D5 10
#define TFT_D6 11
#define TFT_D7 12

#define LOAD_GLCD     // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2    // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4    // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6    // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7    // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8    // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF    // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
```

JimDrewGH, 2021-06-18:
FWIW, adding the define (#define VSPI SPI2_HOST) doesn't work for me with Arduino 1.8.15, ESP32 core v2.0.0-alpha1, and ESP32S2 Dev Module.   I get the same error:

`     spi_host_device_t spi_host = VSPI_HOST;
                                  ^~~~~~~~~
`

Bodmer, 2021-11-18:
The library now supports the ESP32 S2, this has been tested with the Arduino ESP32 board package 2.0.1.

A new setup file example has been added:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70_ESP32_S2_ILI9341.h