Bodmer/TFT_eSPI #1187: ILI9488 (?) driver issue, right driver

handmade0octopus, 2021-05-13:
``Hello.

I have been using ILI9341 2.8" for quite a while now. Wanted to upgrade 3.5" ILI9488.

I tried lots of different things, configurations, even ILI9481, ILI9486 and other drivers but this screen (marketed as ILI9488) doesn't work at all. I tried different SPI speeds (even very slow ones) but no change. Seems like driver issue?

I use this pinout:
```
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   32 
#define TFT_DC    21 
#define TFT_RST   26 

#define TOUCH_CS 22
```

Of course I tried demo and there are strange things happening too.

Any idea [what screen it is if not ILI9488](https://www.ebay.co.uk/itm/265089700228)? Bought from 2 different sources.

Here is [screen with the issue](https://i.imgur.com/syDkh9p.jpg) <===

Bodmer, 2021-05-13:
Run the Read_User_Setup sketch and check the output. Post output here.

handmade0octopus, 2021-05-13:
[code]
TFT_eSPI ver = 2.3.61
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 32
TFT_DC   = GPIO 21
TFT_RST  = GPIO 26
TOUCH_CS = GPIO 22

Font GLCD   loaded

Display SPI frequency = 5.00
Touch SPI frequency   = 2.50
[/code]

Bodmer, 2021-05-14:
I have a visually identical display and the good news is that it is working fine when connected to an ESP32. This means your display should also work.  Here is my setup file:
```
#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
#define TFT_SDA_READ

//#define TFT_BL   32            // LED back-light control pin
//#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)

#define TOUCH_CS 21     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.

#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY       40000000
#define SPI_READ_FREQUENCY  16000000
#define SPI_TOUCH_FREQUENCY  2500000
```
This is the Read_User_Setup output for reference with the latest version of TFT_eSPI:
```
TFT_eSPI ver = 2.3.66
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 21

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50

```

handmade0octopus, 2021-05-21:
I see you using 2.3.66 version of the library. I even tried beta 2.4.0 and nothing.. Same thing you can see video of UTFT_Demo_480x320, maybe this will give you some more information?

https://streamable.com/4kjn1d

I even changed config (on video its 5 MHz SPI clock, 40 MHz is same just faster). I even commented out 99% of setup and tried to get rid of TFT_MISO and TOUCH_CS - but no change (I did that because I have MISO interconnected on PCB with touch MISO but would that cause issues if its not defined at all?).

It's very strange as on same setup ILI9341 is working the same. I am using Arduino ESP32 1.0.5 version.

Are you able to check if exactly same pinout doesn't cause issues? Maybe there is a bug when using pinout like this?

Also both of the screens I have (two different sources) are acting exactly the same strange way.

```
TFT_eSPI ver = 2.4.0
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 32
TFT_DC   = GPIO 21
TFT_RST  = GPIO 26

Font GLCD   loaded

Display SPI frequency = 40.00
```

Bodmer, 2021-05-21:
One possibility is that you have replaced the User_Setup_Select.h file with an old version. Make sure it has these lines 130-135:
https://github.com/Bodmer/TFT_eSPI/blob/8efb988cdabe8e5280db041c82d91d15ac885979/User_Setup_Select.h#L130

handmade0octopus, 2021-05-21:
Yes thank you!

That did the trick!

I setup my own user setup there at the beginning and didn't though there would be some changes.

It's working perfectly now! Thank you for your help and great work with this library!

closed 2021-05-21 by handmade0octopus