Bodmer/TFT_eSPI #1188: Pi Pico compile errors

DustinWatts, 2021-05-14:
Hi! 

I'm getting some compile error when compiling for the Raspberry Pi Pico when using the arduino-pico core by earlephilhower.

Board: Raspberry Pi Pico
Screen / driver: ILI9488
arduino-pico core version: 1.2.2
IDE: 1.8.13 (Windows)
TFT_eSPI version: 2.3.66

First one:
`'MbedSPI' does not name a type`, TFT_eSPI_RP2040.c line 12.

Second repeats a number of times (all the same and my guess related):
`'spi was not declared in this scope; did you mean spi0`

It compiles using the Arduino Mbed OS RP2040 core, but that doesn't seem to work with the ILI9488 (black screen with a bunch of thin lines. So I would like to use the arduino-pico core.

Thanks in advance!

Bodmer, 2021-05-14:
Ah, yes, I had forgotten that I had not added the board package switch code. I will look into that.

The ILI9488 requires 3byte (18bit colour) transfers and although the code for this is present from the generic support it has not been tested.  I will set up a breadboard and get it working.

Incidentally, I have just been using one of the ESP32 + TFT combiner PCBs you designed (from PCBWay) with an ILI9488 SPI display and it is working fine.

DustinWatts, 2021-05-14:
I thought it was something like that, good to hear that is under investigation :)

And also good to hear the ESP32 + TFT Combiner is working for you! I'm planning to make something similar for the Pi Pico.

Bodmer, 2021-05-15:
I have updated the library on Gihub so it now works with the ILI9488. The library now also works with either Arduino or Earle Philhower's board package without any edits.

The ILI9488 RP2040 code has been optimised to boost perfromance as well.

Minor changes for the ESP32 have also been made.

closed 2021-05-15 by Bodmer

DustinWatts, 2021-05-16:
[image]

Works great with the with Arduino!

Unfortunately with Earle Philhower's I just get a white screen for now. Btw... using TFT_Rainbow480 as a test. Can you tell me what version of Earle Philhower's core you use for testing? Also your user_setup.h?

Bodmer, 2021-05-16:
OK, this may be down to the SPI pins being moved to the Pico defined defaults or a recent bug fix:
https://github.com/earlephilhower/arduino-pico/pull/135
 I will investigate.

Bodmer, 2021-05-16:
I just checked an I am using 1.0.3 or Earle's board package. I will upgrade to latest and see what changes may be needed.

Bodmer, 2021-05-16:
I have updated the library to override the default SPI pins when using Earle's board package, so now it should work fine with any any of the allowed RP2040 data sheet SPI pin combinations.

DustinWatts, 2021-05-16:
[image]

Very nice work! Works like a charm!

Bodmer, 2021-05-16:
Mine works at 62.5MHz but this may not be the case for all displays of this type.

commit 5a6ef1d0 (2021-05-15): Fix #1188

Support debugged for ILI9488 TFT with RP2040 processors.
Minor improvements to ESP32 code.