Bodmer/TFT_eSPI #1191: API to do set-up?

frankcohen, 2021-05-15:
Hi! TFT_eSPI is a wonderful library. I'm using it to hack the EleksTube IPS Nixie clock. It's an ESP32 clock using ST7789 135x240 displays to fake the look of nixie tubes. The clock is for sale at https://bit.ly/3uQ8vRn. The current working hacked firmware is at https://github.com/SmittyHalibut/EleksTubeHAX. My firmware to do Wifi and browser config is at https://github.com/frankcohen/EleksTubeIPSHack.

I want to make it easy for others to hack the clock. I am writing this for the average Arduino IDE user to work with. Explaining the steps to modify User_Setup_Select.h seems complicated. And they will need to change the file each time they install or update TFT_eSPI library.

Is there a way to programmatically configure TFT_eSPI? Something like this:

  tft.begin( TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST, TFT_DRIVER, TFT_SDA_READ, TFT_WIDTH, TFT_HEIGHT,  TFT_SPI, LOAD_GLCD );

These would override the User_Setup_Select.h.

-Frank

Bodmer, 2021-05-15:
Yes, this is possible but the changes to the library would be quite significant due to the way it has evolved and the code size would grow since the compiler will include all config code for every display type.

The objective of doing it via the setup files is that it is quick nand easy to select between different hardware setups and then the compatible examples all run without change.  With PlatformIO you can define the settings in the project ini file.

The simplest way to support users with fixed hardware is to provide a complete replacement User_Setup.h file that suits your hardware design, then users can also run all the examples too.

closed 2021-05-15 by Bodmer

frankcohen, 2021-05-15:
Makes sense to me why you chose the set-up path… for the reasons you enumerated.

For my project I will support the User_Setup.h file for people compiling the firmware from the source. And, I will provide a compiled bin file that they can flash to the ESP32 directly.

Thanks for your quick support!

-Frank