Bodmer/TFT_eSPI #1192: doe's fillscreen(xxx)  real fills the whole screen?

MHz000, 2021-05-17:
using a ST7735 green tab my platformio.ini looks like:
[env:d1]
platform = espressif8266
board = d1
framework = arduino
monitor_speed = 115200
upload_speed = 512000
build_flags = 
	-DUSER_SETUP_LOADED=1
	-DST7735_DRIVER=1	
	-DTFT_WIDTH=240
	-DTFT_HEIGHT=320
	-DST7735_GREENTAB3=1    << tried all possible setings  from no one to this one
	-DTFT_RST=-1
	-DTFT_CS=15
	-DTFT_DC=0
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_FONT8N=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=27000000
lib_deps = bodmer/TFT_eSPI@^2.3.69

and my source code looks like this:
#include <Arduino.h>
#include <SPI.h>
#include <TFT_eSPI.h>            // Hardware-specific library
TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

  int wS;                         // screen width            
  int hS;                         // screen hight

void setup() {
    Serial.begin(115200);
  delay(200);
  Serial.printf("\n\nCompiled from: %s  at: %s %s", __FILE__, __DATE__, __TIME__);
  Serial.println("\nXXXXXXXXXXXXXX");

  tft.init();
  tft.setRotation(0);                

  int wS = tft.width();
  int hS = tft.height();
  Serial.printf("Screen size %u * %u", wS, hS);
  tft.fillRect(0,0,hS,wS,TFT_RED);
  delay(1000);
  tft.fillScreen(TFT_BLACK);
  //tft.fillScreen(TFT_RED);
  //tft.fillScreen(TFT_GREEN);
  //tft.fillScreen(TFT_BLUE);
  //tft.fillScreen(TFT_BLACK);
  //tft.fillScreen(TFT_GREY);
}
void loop() {
  // put your main code here, to run repeatedly:
}

screen dimensions reported correctly by program as: Screen size 240 * 320
tft.fillRect(0,0,hS,wS,TFT_RED); fills the whole screen as expected
however tft.fillScreen(TFT_BLACK) fills just the right third of the screen  => see picture

What is wrong with my settings?
Thank-you

Btw:  the  TFT_eSPI/examples/320 x 240/TFT_Clock/TFT_Clock.ino does not respect the width and hight settings see picture. Maybe the same reason as with fillScreen. Again what is wrong with my settings?

[image]

[image]

Bodmer, 2021-05-17:
This:
```
-DTFT_WIDTH=240
-DTFT_HEIGHT=320
```
tells the  library your screen size is 240 x 320. The ST7735 screen is 128 x 160 so the library is expecting a larger pixel count screen to be connected.

Either leave out those parameters so the library uses a default size, or set to:
```
-DTFT_WIDTH=128
-DTFT_HEIGHT=160
```

closed 2021-05-17 by Bodmer

MHz000, 2021-05-17:
my fault 😖 all problems solved 😊
thank-you so much!