Bodmer/TFT_eSPI #1198: class TFT_eSprite has no member named 'loadFont'

DRMartin4321, 2021-05-19:
error compiling Animated_dial.ino

in function 'void setup()'
Animated_dial:84:7 error 'class TFTeSprite' has no member named 'loadFont'
spr.loadFont(AA_FONT_LARGE);

Bodmer, 2021-05-21:
Make sure you have smooth fonts enabled in the setup file See example here:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L305

Bodmer, 2021-05-24:
I assume this is solved by user now, so closing.

closed 2021-05-24 by Bodmer

ats3788, 2025-01-06:
Can't Compile SmoothFont Example
 error: 'class TFT_eSPI' has no member named 'showFont'; did you mean 'gfxFont'?

The Smooth_font.h Starts with this can't work

>  // Coded by Bodmer 10/2/18, see license in root directory.
>  // This is part of the TFT_eSPI class and is associated with anti-aliased font functions
> 
>  public:
> 
>   // These are for the new anti-aliased fonts
>   void     loadFont(const uint8_t array[]);
> #ifdef FONT_FS_AVAILABLE
>   void     loadFont(String fontName, fs::FS &ffs);
> #endif
>   void     loadFont(String fontName, bool flash = true);
>   void     unloadFont( void );
>   bool     getUnicodeIndex(uint16_t unicode, uint16_t *index);
> 
>   virtual void drawGlyph(uint16_t code);
> 
>   void     showFont(uint32_t td);
> 
>  // This is for the whole font
>   typedef struct
>   {
>     const uint8_t* gArray;           //array start pointer
>     uint16_t gCount;                 // Total number of characters
>     uint16_t yAdvance;               // Line advance
>     uint16_t spaceWidth;             // Width of a space character
>     int16_t  ascent;                 // Height of top of 'd' above baseline, other characters may be taller
>     int16_t  descent;                // Offset to bottom of 'p', other characters may have a larger descent
>     uint16_t maxAscent;              // Maximum ascent found in font
>     uint16_t maxDescent;             // Maximum descent found in font
>   } fontMetrics;