Bodmer/TFT_eSPI #1200: undefined reference to `TFT_eSPI_Button::initButton

3mc2, 2021-05-21:
I am attempting to build the sample to an ESP32 Dev Board.  In total, I get the following error.   I tried including the button.h file but that created redefinition errors.  

Any advice would be appreciated.

sketch\sketch_may21a.ino.cpp.o:(.literal._Z10drawKeypadv+0x24): undefined reference to `TFT_eSPI_Button::initButton(TFT_eSPI*, short, short, unsigned short, unsigned short, unsigned short, unsigned short, unsigned short, char*, unsigned char)'
sketch\sketch_may21a.ino.cpp.o:(.literal._Z10drawKeypadv+0x28): undefined reference to `TFT_eSPI_Button::drawButton(bool, String)'
sketch\sketch_may21a.ino.cpp.o:(.literal.startup._GLOBAL__sub_I_tft+0x4): undefined reference to `TFT_eSPI_Button::TFT_eSPI_Button()'
sketch\sketch_may21a.ino.cpp.o: In function `drawKeypad()':
C:\Users\rimbr\OneDrive\Documents\Arduino\sketch_may21a/sketch_may21a.ino:154: undefined reference to `TFT_eSPI_Button::initButton(TFT_eSPI*, short, short, unsigned short, unsigned short, unsigned short, unsigned short, unsigned short, char*, unsigned char)'
C:\Users\rimbr\OneDrive\Documents\Arduino\sketch_may21a/sketch_may21a.ino:155: undefined reference to `TFT_eSPI_Button::drawButton(bool, String)'
sketch\sketch_may21a.ino.cpp.o: In function `_GLOBAL__sub_I_tft':
C:\Users\rimbr\OneDrive\Documents\Arduino\sketch_may21a/sketch_may21a.ino:74: undefined reference to `TFT_eSPI_Button::TFT_eSPI_Button()'
collect2.exe: error: ld returned 1 exit status
exit status 1

Bodmer, 2021-05-22:
Try the[ example here ](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/Keypad_240x320/Keypad_240x320.ino)and build on that.

Bodmer, 2021-05-24:
I assume this is solved by user now, so closing.

closed 2021-05-24 by Bodmer

Bodmer, 2022-01-08:
I suspect this happened because there was a class link missing.

It has been fixed in v2.4.23