Bodmer/TFT_eSPI #1204: Error al compilar RPI_PICO con display TFT ILI9225

Luuciano27, 2021-05-23:
Hola, estoy usando un raspberry pi pico con un display TFT 2.2 basado en el controlador ILI9225. Cuando compilo me da el siguiente error:
```
In file included from C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:26:
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:17:61: error: 'TFT_SCLK' was not declared in this scope
   17 |     SPIClassRP2040 spi = SPIClassRP2040(spi0, TFT_MISO, -1, TFT_SCLK, TFT_MOSI);
      |                                                             ^~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:17:71: error: 'TFT_MOSI' was not declared in this scope; did you mean 'TFT_MISO'?
   17 |     SPIClassRP2040 spi = SPIClassRP2040(spi0, TFT_MISO, -1, TFT_SCLK, TFT_MOSI);
      |                                                                       ^~~~~~~~
      |                                                                       TFT_MISO
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:23:3: error: 'dma_channel_config' does not name a type
   23 |   dma_channel_config dma_tx_config;
      |   ^~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'bool TFT_eSPI::dmaBusy()':
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:305:7: error: 'dma_channel_is_busy' was not declared in this scope
  305 |   if (dma_channel_is_busy(dma_tx_channel)) return true;
      |       ^~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'void TFT_eSPI::dmaWait()':
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:317:10: error: 'dma_channel_is_busy' was not declared in this scope
  317 |   while (dma_channel_is_busy(dma_tx_channel));
      |          ^~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'void TFT_eSPI::pushPixelsDMA(uint16_t*, uint32_t)':
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:332:29: error: 'dma_tx_config' was not declared in this scope
  332 |   channel_config_set_bswap(&dma_tx_config, !_swapBytes);
      |                             ^~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:332:3: error: 'channel_config_set_bswap' was not declared in this scope
  332 |   channel_config_set_bswap(&dma_tx_config, !_swapBytes);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:333:3: error: 'dma_channel_configure' was not declared in this scope
  333 |   dma_channel_configure(dma_tx_channel, &dma_tx_config, &spi_get_hw(spi0)->dr, (uint16_t*)image, len, true);
      |   ^~~~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'void TFT_eSPI::pushImageDMA(int32_t, int32_t, int32_t, int32_t, uint16_t*, uint16_t*)':
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:380:29: error: 'dma_tx_config' was not declared in this scope
  380 |   channel_config_set_bswap(&dma_tx_config, !_swapBytes);
      |                             ^~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:380:3: error: 'channel_config_set_bswap' was not declared in this scope
  380 |   channel_config_set_bswap(&dma_tx_config, !_swapBytes);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:381:3: error: 'dma_channel_configure' was not declared in this scope
  381 |   dma_channel_configure(dma_tx_channel, &dma_tx_config, &spi_get_hw(spi0)->dr, (uint16_t*)buffer, len, true);
      |   ^~~~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'bool TFT_eSPI::initDMA(bool)':
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:393:20: error: 'dma_claim_unused_channel' was not declared in this scope
  393 |   dma_tx_channel = dma_claim_unused_channel(true);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:394:3: error: 'dma_tx_config' was not declared in this scope
  394 |   dma_tx_config = dma_channel_get_default_config(dma_tx_channel);
      |   ^~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:394:19: error: 'dma_channel_get_default_config' was not declared in this scope
  394 |   dma_tx_config = dma_channel_get_default_config(dma_tx_channel);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:396:57: error: 'DMA_SIZE_16' was not declared in this scope; did you mean 'DMA_IRQ_1'?
  396 |   channel_config_set_transfer_data_size(&dma_tx_config, DMA_SIZE_16);
      |                                                         ^~~~~~~~~~~
      |                                                         DMA_IRQ_1
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:396:3: error: 'channel_config_set_transfer_data_size' was not declared in this scope
  396 |   channel_config_set_transfer_data_size(&dma_tx_config, DMA_SIZE_16);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:397:3: error: 'channel_config_set_dreq' was not declared in this scope
  397 |   channel_config_set_dreq(&dma_tx_config, spi_get_index(spi0) ? DREQ_SPI1_TX : DREQ_SPI0_TX);
      |   ^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'void TFT_eSPI::deInitDMA()':
C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:410:3: error: 'dma_channel_unclaim' was not declared in this scope
  410 |   dma_channel_unclaim(dma_tx_channel);
      |   ^~~~~~~~~~~~~~~~~~~
exit status 1
Error compilando para la tarjeta Raspberry Pi Pico.

```
Espero puedas ayudarme, gracias.

Bodmer, 2021-05-23:
The first error tells me you are using Earle Philhower's board package which is OK. It also tells me that TFT_SCLK pin has not been defined.  This is probably because you have not eddited the User_Setup_Select.h file in the library folder. Comment out the line:
```
//#include <User_Setup.h>           // Default setup is root library folder
```
Uncomment the line:
```
#include <User_Setups/Setup60_RP2040_ILI9341.h>    // Setup file for Raspberry Pi Pico with SPI ILI9341
```
The sketch should then compile OK. You may need to edit the User_Setups/Setup60_RP2040_ILI9341.h file to suit your hardware configuration.

Luuciano27, 2021-05-23:
Hola @Bodmer , gracias por tu respuesta, intente lo que me sugeriste pero si fallando al compilar, dice lo siguiente:
Arduino:1.8.13 (Windows 10), Tarjeta:"Raspberry Pi Pico, 2MB (no FS), 125 MHz, Disabled, None, Pico SDK"

```
In file included from C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:26:

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:23:3: error: 'dma_channel_config' does not name a type

   23 |   dma_channel_config dma_tx_config;

      |   ^~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'bool TFT_eSPI::dmaBusy()':

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:305:7: error: 'dma_channel_is_busy' was not declared in this scope

  305 |   if (dma_channel_is_busy(dma_tx_channel)) return true;

      |       ^~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'void TFT_eSPI::dmaWait()':

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:317:10: error: 'dma_channel_is_busy' was not declared in this scope

  317 |   while (dma_channel_is_busy(dma_tx_channel));

      |          ^~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'void TFT_eSPI::pushPixelsDMA(uint16_t*, uint32_t)':

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:332:29: error: 'dma_tx_config' was not declared in this scope

  332 |   channel_config_set_bswap(&dma_tx_config, !_swapBytes);

      |                             ^~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:332:3: error: 'channel_config_set_bswap' was not declared in this scope

  332 |   channel_config_set_bswap(&dma_tx_config, !_swapBytes);

      |   ^~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:333:3: error: 'dma_channel_configure' was not declared in this scope

  333 |   dma_channel_configure(dma_tx_channel, &dma_tx_config, &spi_get_hw(spi0)->dr, (uint16_t*)image, len, true);

      |   ^~~~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'void TFT_eSPI::pushImageDMA(int32_t, int32_t, int32_t, int32_t, uint16_t*, uint16_t*)':

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:380:29: error: 'dma_tx_config' was not declared in this scope

  380 |   channel_config_set_bswap(&dma_tx_config, !_swapBytes);

      |                             ^~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:380:3: error: 'channel_config_set_bswap' was not declared in this scope

  380 |   channel_config_set_bswap(&dma_tx_config, !_swapBytes);

      |   ^~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:381:3: error: 'dma_channel_configure' was not declared in this scope

  381 |   dma_channel_configure(dma_tx_channel, &dma_tx_config, &spi_get_hw(spi0)->dr, (uint16_t*)buffer, len, true);

      |   ^~~~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'bool TFT_eSPI::initDMA(bool)':

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:393:20: error: 'dma_claim_unused_channel' was not declared in this scope

  393 |   dma_tx_channel = dma_claim_unused_channel(true);

      |                    ^~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:394:3: error: 'dma_tx_config' was not declared in this scope

  394 |   dma_tx_config = dma_channel_get_default_config(dma_tx_channel);

      |   ^~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:394:19: error: 'dma_channel_get_default_config' was not declared in this scope

  394 |   dma_tx_config = dma_channel_get_default_config(dma_tx_channel);

      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:396:57: error: 'DMA_SIZE_16' was not declared in this scope; did you mean 'DMA_IRQ_1'?

  396 |   channel_config_set_transfer_data_size(&dma_tx_config, DMA_SIZE_16);

      |                                                         ^~~~~~~~~~~

      |                                                         DMA_IRQ_1

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:396:3: error: 'channel_config_set_transfer_data_size' was not declared in this scope

  396 |   channel_config_set_transfer_data_size(&dma_tx_config, DMA_SIZE_16);

      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:397:3: error: 'channel_config_set_dreq' was not declared in this scope

  397 |   channel_config_set_dreq(&dma_tx_config, spi_get_index(spi0) ? DREQ_SPI1_TX : DREQ_SPI0_TX);

      |   ^~~~~~~~~~~~~~~~~~~~~~~

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c: In member function 'void TFT_eSPI::deInitDMA()':

C:\Users\Luciano\Documents\Arduino\libraries\TFT_eSPI-master\Processors/TFT_eSPI_RP2040.c:410:3: error: 'dma_channel_unclaim' was not declared in this scope

  410 |   dma_channel_unclaim(dma_tx_channel);

      |   ^~~~~~~~~~~~~~~~~~~

exit status 1

Error compilando para la tarjeta Raspberry Pi Pico.
```
Quizás me falte instalar algún archivo?

rcla, 2021-05-23:
Maybe inside the Processors/TFT_eSPI_RP2040.h file,

Uncomment the line:
`#include "hardware/dma.h"`

cgidzinski, 2021-05-24:
Just adding some more info as I also had the same issue using the "Arduino Mbed OS RP2040 Boards -> Raspberry Pi Pico" and it was working prior to https://github.com/Bodmer/TFT_eSPI/commit/5a6ef1d05d19e2d4277b1cb8d0a0e7034aefad09 . After un-commenting `#include "hardware/dma.h"` as mentioned above it compiles successfully on both DMA and non DMA specific examples

Luuciano27, 2021-05-24:
Agradezco mucho su ayuda, funciono bien lo que me recomendaron @rcla, @cgidzinski , ya compila bien el código.
Ahora tengo otro problema, lo configure para mi display ILI9225 y solo muestra lineas, no se muestra correctamente, que podría ser?
Las configuraciones que realicé son en el archivo `Setup60_RP2040_ILI9341`, comente la linea del controlador ILI9341 y eliminé el comentario de `#define ILI9225_DRIVER`.
Configure los pines del SPI de la siguiente manera: 
```
// For the Pico use these #define lines
#define TFT_MISO  99
#define TFT_MOSI  3
#define TFT_SCLK  2
#define TFT_CS   5  // Chip select control pin
#define TFT_DC   7  // Data Command control pin
#define TFT_RST  6  // Reset pin (could connect to Arduino RESET pin)
//#define TFT_BL     // LED back-light
```

Luuciano27, 2021-05-24:
Este es el boceto de ejemplo que estoy utilizando:
```
/*
 An example analogue clock using a TFT LCD screen to show the time
 use of some of the drawing commands with the library.

 For a more accurate clock, it would be better to use the RTClib library.
 But this is just a demo. 
 
 This sketch uses font 4 only.

 Make sure all the display driver and pin connections are correct by
 editing the User_Setup.h file in the TFT_eSPI library folder.

 #########################################################################
 ###### DON'T FORGET TO UPDATE THE User_Setup.h FILE IN THE LIBRARY ######
 #########################################################################
 
 Based on a sketch by Gilchrist 6/2/2014 1.0
 */

#include <SPI.h>
#include <TFT_eSPI.h> // Hardware-specific library

#define TFT_GREY 0x5AEB

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

float sx = 0, sy = 1, mx = 1, my = 0, hx = -1, hy = 0;    // Saved H, M, S x & y multipliers
float sdeg=0, mdeg=0, hdeg=0;
uint16_t osx=120, osy=120, omx=120, omy=120, ohx=120, ohy=120;  // Saved H, M, S x & y coords
uint16_t x0=0, x1=0, yy0=0, yy1=0;
uint32_t targetTime = 0;                    // for next 1 second timeout

static uint8_t conv2d(const char* p); // Forward declaration needed for IDE 1.6.x
uint8_t hh=conv2d(__TIME__), mm=conv2d(__TIME__+3), ss=conv2d(__TIME__+6);  // Get H, M, S from compile time

bool initial = 1;

void setup(void) {
  tft.init();
  tft.setRotation(0);
  
  //tft.fillScreen(TFT_BLACK);
  //tft.fillScreen(TFT_RED);
  //tft.fillScreen(TFT_GREEN);
  //tft.fillScreen(TFT_BLUE);
  //tft.fillScreen(TFT_BLACK);
  tft.fillScreen(TFT_GREY);
  
  tft.setTextColor(TFT_WHITE, TFT_GREY);  // Adding a background colour erases previous text automatically
  
  // Draw clock face
  tft.fillCircle(120, 120, 118, TFT_GREEN);
  tft.fillCircle(120, 120, 110, TFT_BLACK);

  // Draw 12 lines
  for(int i = 0; i<360; i+= 30) {
    sx = cos((i-90)*0.0174532925);
    sy = sin((i-90)*0.0174532925);
    x0 = sx*114+120;
    yy0 = sy*114+120;
    x1 = sx*100+120;
    yy1 = sy*100+120;

    tft.drawLine(x0, yy0, x1, yy1, TFT_GREEN);
  }

  // Draw 60 dots
  for(int i = 0; i<360; i+= 6) {
    sx = cos((i-90)*0.0174532925);
    sy = sin((i-90)*0.0174532925);
    x0 = sx*102+120;
    yy0 = sy*102+120;
    // Draw minute markers
    tft.drawPixel(x0, yy0, TFT_WHITE);
    
    // Draw main quadrant dots
    if(i==0 || i==180) tft.fillCircle(x0, yy0, 2, TFT_WHITE);
    if(i==90 || i==270) tft.fillCircle(x0, yy0, 2, TFT_WHITE);
  }

  tft.fillCircle(120, 121, 3, TFT_WHITE);

  // Draw text at position 120,260 using fonts 4
  // Only font numbers 2,4,6,7 are valid. Font 6 only contains characters [space] 0 1 2 3 4 5 6 7 8 9 : . - a p m
  // Font 7 is a 7 segment font and only contains characters [space] 0 1 2 3 4 5 6 7 8 9 : .
  tft.drawCentreString("Time flies",120,260,4);

  targetTime = millis() + 1000; 
}

void loop() {
  if (targetTime < millis()) {
    targetTime += 1000;
    ss++;              // Advance second
    if (ss==60) {
      ss=0;
      mm++;            // Advance minute
      if(mm>59) {
        mm=0;
        hh++;          // Advance hour
        if (hh>23) {
          hh=0;
        }
      }
    }

    // Pre-compute hand degrees, x & y coords for a fast screen update
    sdeg = ss*6;                  // 0-59 -> 0-354
    mdeg = mm*6+sdeg*0.01666667;  // 0-59 -> 0-360 - includes seconds
    hdeg = hh*30+mdeg*0.0833333;  // 0-11 -> 0-360 - includes minutes and seconds
    hx = cos((hdeg-90)*0.0174532925);    
    hy = sin((hdeg-90)*0.0174532925);
    mx = cos((mdeg-90)*0.0174532925);    
    my = sin((mdeg-90)*0.0174532925);
    sx = cos((sdeg-90)*0.0174532925);    
    sy = sin((sdeg-90)*0.0174532925);

    if (ss==0 || initial) {
      initial = 0;
      // Erase hour and minute hand positions every minute
      tft.drawLine(ohx, ohy, 120, 121, TFT_BLACK);
      ohx = hx*62+121;    
      ohy = hy*62+121;
      tft.drawLine(omx, omy, 120, 121, TFT_BLACK);
      omx = mx*84+120;    
      omy = my*84+121;
    }

      // Redraw new hand positions, hour and minute hands not erased here to avoid flicker
      tft.drawLine(osx, osy, 120, 121, TFT_BLACK);
      osx = sx*90+121;    
      osy = sy*90+121;
      tft.drawLine(osx, osy, 120, 121, TFT_RED);
      tft.drawLine(ohx, ohy, 120, 121, TFT_WHITE);
      tft.drawLine(omx, omy, 120, 121, TFT_WHITE);
      tft.drawLine(osx, osy, 120, 121, TFT_RED);

    tft.fillCircle(120, 121, 3, TFT_RED);
  }
}

static uint8_t conv2d(const char* p) {
  uint8_t v = 0;
  if ('0' <= *p && *p <= '9')
    v = *p - '0';
  return 10 * v + *++p - '0';
}
```

Bodmer, 2021-05-24:
Yep, that include should not have been cmmented out. Now fixed.

The include may cause issues with very erly versions of the RP2040 board packages so please use latest versions.

commit c7a3f464 (2021-05-24): Fix #1204

This include can now be used with either the latest versions of Arduino official or Earle Philhower's board package.
closed 2021-05-24 by Bodmer