Bodmer/TFT_eSPI #1212: Example 'TFT_Meter_linear' - scale down

dl5dla, 2021-05-24:
I would like to use the meter just on one quarter of the display (ILI9341). I guess in general I can do this by reducing M_SIZE, but the font size still remains the same as before. As far as I understood the font size can be changed within the lines 
```
  case -2: tft.drawCentreString("0", x0, y0 - 12, **2**); 
  ...
```
but it seems that I can only make the font bigger, not smaller. I am wondering whether the used font is already the smallest possible (?).
Thank you for any tipps or clarification.

Bodmer, 2021-05-24:
You can try the GLCD font (font 1) but you may have to adjust the x,y positions of the values:
```
case -2: tft.drawCentreString("0", x0, y0 - 12, 1); break;
```

closed 2021-05-24 by Bodmer

dl5dla, 2021-05-25:
Thanks for your answer. I changed the font to "1" for the scale point "0" and "25" but now they are even a little bit bigger.

[image]

Bodmer, 2021-05-25:
This happens because you have changed the example to load a free font, so that font replaces the GLCD font. Use this line before printing the scale to select the GLCD font.
```
tft.setTextFont(1);
```
There is also a smaller Tomthumb free font.

dl5dla, 2021-05-27:
Thank al ot - that solved it.