Bodmer/TFT_eSPI #1213: Partial fix Raspberry Pi Pico with RPI Display Type

rcla, 2021-05-26:
I have been trying to get the Raspberry Pi Pico to work with the Waveshare 3.5inch RPi LCD (A) ILI9486 display.
More details in:
[Discussion](https://github.com/Bodmer/TFT_eSPI/discussions/1111#discussioncomment-700104)

Then what I did was connect a Teensy with that display and using its library and it worked!

So I started to investigate and after many tests and experiments a comment caught my attention:
`// RPi TFT type always needs 16 bit transfers`
[RPI_DISPLAY_TYPE](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_RP2040.h#L159)

So inside the "**setWindow**" function, modify this part:
[setWindow](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L3095)

```
    #if !defined (SPI_18BIT_DRIVER)
      #if  defined (RPI_DISPLAY_TYPE) // RPi TFT type always needs 16 bit transfers
      	spi_set_format(spi0,  16, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST);
      #else
        spi_set_format(spi0,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST);
      #endif
    #endif
```
And now it works!
[image]

Tried doing the same with other works, but I know it takes more work to really be a solution.

If I run the example "TFT_Rainbow480" some letters are not seen or "garbage" is displayed.
And in the example "Bouncy_Circles" I only get a black screen.

Bodmer, 2021-05-27:
At the moment the RP2040 has not been tested with any RPi displays (or any parallel 8 bit displays). I have other priority projects at the moment but will add this to the TODO list.

Bodmer, 2021-11-05:
This should now be fixed in the latest version 2.3.73.

Try it and post back if you have a problem.

closed 2021-11-05 by Bodmer