Bodmer/TFT_eSPI #1215: Support for ILI9328 and HX8357D

hotteshen, 2021-05-27:
I could use this library for 3.5 inch TFT LCDs with ILI9381 driver.

I also want to use 2.8 inch Arduino TFT LCD Shields. I bought two different shield.
* [TFT Shield with HX8347D](https://de.aliexpress.com/item/32831754174.html)
* [TFT Shield with ILI9325/9328](https://wiki.keyestudio.com/Ks0257_keyestudio_2.8_Inch_TFT_LCD_Shield)

<img srd="https://wiki.keyestudio.com/images/a/a3/%E5%9B%BE%E7%89%871-_ks0257.png" style="width: 40%;">

I am seeing those driver ICs are not supported by TFT_eSPI yet.

As these TFT LCD shields are very cheap, I'd like if this library works with those displays.
Any plan to support these?

Bodmer, 2021-05-27:
It is possible to add new drivers to the library buit without a display to test them on this tends to be a rather troublesome activity for me to do.  Maybe someone else can help.

hotteshen, 2021-06-13:
For a **temporal** solution till TFT_eSPI supports those driver chips, [MCUFRIEND_kbv](https://github.com/prenticedavid/MCUFRIEND_kbv) can be used.
It supports all those above (HX8347D, ILI9325/9328).

But it's slower than TFT_espi.
And the worse thing for me was that I couldn't make it work with [LVGL](https://github.com/lvgl/lvgl)

Rocosso, 2021-07-05:
@Bodmer how can I send you a TFT 8347i, for help you to help another people?

hotteshen, 2021-07-05:
I also have HX8347 and ILI9328 displays unused.
After a lot of tries, I decided to use ILI9341 LCD and already designed my product with it.
If possible, I also would like to ship my HX8347 and ILI9328 LCDs to your address. (Of course if @Bodmer has time to work with those)

Rocosso, 2021-07-05:
@Bodmer I don't  know  to program in low level, with many dificulties i try to learn how to program, I want to help the community, please let know if we can send you this screens, or who can receive it for make better the TFT_eSPI library.

Thanks for sharing your knowledge, your pattience, your time, your strength and your affords.

amitgithub1, 2022-03-01:
> 

Somehow it works for me with ILI9328 LCD driver (as detected by adafruit LCD library on uno), on NodeMCU 32s
i tried a lot of things but finally the below two changes made it work:
1) using 3.3v instead of 5v
2) Uncommenting the first "#define TFT_PARALLEL_8_BIT" line number 33 also

If anything, do ask as i am working on it now ..

amitgithub1, 2022-03-01:
There doesn't seem to be any 5 to 3.3v id behind the screen pcb.
Any suggestion where i should send the pic for reference?

A couple examples are working except touch based, which i am not able to run seems parallel lcd do not have touch support here.

amitgithub1, 2022-03-01:
> In my experiments, ILI9328 LCDs did not work with ILI9341 drivers. Could you share a close picture of your LCD module?

Hey, I am very sorry, there seem to be an confusion at my side, i had ordered two LCD together, again rechecking i find that both are different driver. Will update if anything works.