Bodmer/TFT_eSPI #1217: doesnt work for pi pico mbed or earle

ACencernedMember, 2021-05-27:
i compile it throws errors on either. adafruit gfx one thats slow works sadly. i address some of the errors and it throws more errors.

is there another library or even separate plugin i need to make this work? no shield i wired the pi pico directly to the screen. i have only used arduino and never micropython or circuit python on this pi pico

i know its alot of work but maybe you could separate out the different screens so that there isnt more code than needed.i tried but it still throws errors

Bodmer, 2021-05-29:
I suspect you have not eddited the setup files within the library correctly. 

Comment out (add // to line start) [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L22). And uncomment (remove //) [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L81), then recompile and upload.

Bodmer, 2021-06-05:
Assumed solved.

closed 2021-06-05 by Bodmer