Bodmer/TFT_eSPI #1220: TTGO T1-Display going blank

pravinkadam76, 2021-05-29:
ESP32 TTGO T - Display going blank with this library, interesting this is that its not working with the old library version too! I have tried it on different laptop with older version of TFT_eSPI library, the program complies fine, but the screen is blank...looks like something to do with the backlight pin or value

Bodmer, 2021-05-29:
You have not eddited the setup files correctly. The T display is suported, use versions of User_Setup.h and User_Setup_Select.h with the orginal library files (do not try to transplant from other library copies), then comment out (add // to line start) [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L22). And uncomment (remove //) [this line,](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L53) then recompile and upload.

pravinkadam76, 2021-05-30:
Thanks for the quick reply! Appreciate it!

Cheers!
Pravin

spiff72, 2021-06-03:
I'm not sure, but I may be running into something similar.  I have a usermod for the aircoookie WLED project (https://github.com/Aircoookie/WLED/tree/master/usermods/TTGO-T-Display) that someone just informed me is creating a blank display when they try it out.  I just went and cloned a fresh copy of the WLED main branch to see if I was working for me, but I get the same blank display.

I am compiling with VS Code/PlatformIO on a Mac, and I let it grab the TFT_eSPI library at compile time.  I commented out the correct lines as I described above (as I have been for a few years without issue), but still a blank screen.

Any ideas?

Bodmer, 2021-06-05:
@spiff72 
Do the TFT_ePI library examples work? If yes, then I am unable to help but the WLED authors may be able to advise.
Run the Read_User_Setup example to check the compiler is picking up the right settings.

Bodmer, 2021-06-05:
@pravinkadam76
Closing as I assume your issue is solved.

closed 2021-06-05 by Bodmer

spiff72, 2021-06-05:
I will try and get back to you.  thanks!

spiff72, 2021-06-05:
OK- I am a bit rusty with Arduino IDE, but I managed to run that example sketch (twice).  The first time I ran it I didn't change the user_select_setup file, and the second time I edited the file in the arduino library folder to comment out the first line and uncommented the line for the TTGO display.  Here is the output from the serial monitor for both of those runs:

```
[code]
TFT_eSPI ver = 2.3.70
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 0
TFT_RST  = GPIO 2

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]
  ⸮ ⸮ ⸮⸮ ⸮  ⸮⸮⸮  ⸮  ⸮ ⸮⸮ ⸮    ⸮   ⸮
[code]
TFT_eSPI ver = 2.3.70
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 135
Display height = 240

R0 x offset = 52
R0 y offset = 40
MOSI    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 5
TFT_DC   = GPIO 16
TFT_RST  = GPIO 23

TFT_BL           = GPIO 4
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
[/code]
```

spiff72, 2021-06-05:
Also, the screen on the TTGO module is showing a disported "snow"-like image after running this.  Is this normal?  I don't see any commands in the sketch to write to the display...

spiff72, 2021-06-05:
I also ran the digital clock example from the lowest resolution example folder, and that one works correctly (just doesn't fill the screen).

spiff72, 2021-06-05:
Unfortunately, I am unable to figure out how to do this same example work from within VS Code, so I am running these a bit differently than I have been with the WLED project (which I only do in VS code).