Bodmer/TFT_eSPI #1221: TFT ili9341 White screen with esp 32

medamine0919, 2021-05-31:
When I try to run my TFT ili9341 2.8 with esp32 it shows white screen 
I have tried all exemples but no luck 
I also tried to change the spi pins number in user setup but nothing changed
Please help

Bodmer, 2021-06-05:
Run the Read_Set_Setup sketch and check the compiler is picking up the correct settings.

medamine0919, 2021-06-05:
sorry for forgetting to give a feedback but i solved that problem by commenting the "CS" in the user_setup file and connecting the CS pin to ground but now i have another problem that the tft is resetting after a while from putting it to work.
any idea how to fix that ??
thanks in advance

Bodmer, 2021-06-05:
The ESP32 has a watchdog time-out of about 1 second. If you sketch does not run through the loop() function in less than 1 second then you may have to add yield((); to your sketch.

The library examples should not cause a watchdog reset unless you are running at a low SPI clock rate.

closed 2021-06-16 by Bodmer