Bodmer/TFT_eSPI #1231: ST7789V - colors washed (like light colors)

slavendam, 2021-06-10:
I have 2 screen. One is ST7789 and other is ST7789V. On ST7789 everything works great, colors look as they should be. 
On ST7789V colors looks like they are washed. If I set BLACK screen, it looks like light black or grey. If I set BLUE, it looks like light blue.
All screens are same so it is not hardware problem.

Is there something in library that I can change to get more contrast?

Thanks!

Bodmer, 2021-06-17:
If you can get the init sequence from the supplier of your V type TFT then I can add it.

You could also try the alternative #define ST7789_2_DRIVER driver, [see here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L49-L50):
```
//#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
```

slavendam, 2021-06-17:
Here are documents they sent to me:
https://www.scribd.com/document/512133947/ST013-01
https://www.scribd.com/document/512133966/ST7789V

I tried setting ST7789_2 but without success.

Thank you for your help!

Bodmer, 2021-11-21:
Unfortunately I do not have that particular display to test but I suspect it needs different bias voltage settings in the init sequence.

closed 2021-11-21 by Bodmer