Bodmer/TFT_eSPI #1237: Does the ILI9488 0xB1 command missing a second data byte?

zapta, 2021-06-19:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9488_Init.h#L70

Command 0XB1, page 220 here https://focuslcds.com/content/ILI9488.pdf

Bodmer, 2021-09-14:
This does not seem to be important for the serial/parallel displays I have. The init code came from a display vendors website but that does not mean it is correct for all ILI9488 based displays.

I suspect the second register defaults to a sensible value.
Are you have a problem with your display?
What second byte values worked for you?
Can you post a link to the display vendors website if it is a new variant that requires thay extra byte?

zapta, 2021-09-15:
Hi,

This is what I am using
https://github.com/zapta/simple_stepper_motor_analyzer/blob/main/platformio/src/display/tft_driver.cpp#L405

I noticed the missing byte when I tried to understand the
initialization sequence based on the data sheet (page 220).
 https://focuslcds.com/content/ILI9488.pdf

BTW, are you interested in adding RPI Pico/RP2040 8 bit parallel mode that
is based on the PIO? It's much faster than SPI.

Bodmer, 2021-11-21:
Yes parallelPIO with DMA is on my todo list but I have not had time to look at this.

closed 2021-11-21 by Bodmer

zapta, 2021-11-21:
You can find ILI9488 PIO example here

PIO
https://github.com/zapta/simple_stepper_motor_analyzer/blob/main/platformio/src/display/tft_driver.pio

C driver
https://github.com/zapta/simple_stepper_motor_analyzer/blob/main/platformio/src/display/tft_driver.cpp#L473

Bodmer, 2021-11-24:
Thanks I will have a look a this.