Bodmer/TFT_eSPI #1240: SD card + TFT + touch not working with FreeRTOS Tasks

HKpro2090, 2021-06-22:
I am using an external sd card module, TFT screen with touch. I created two tasks,  one to see if the screen is being touched and return it is coordinates and the other one to displaying time from the font used from the SD card. I have used the Same MISO pin for the SD card and touch. The screen is showing when I switch off the touch screen task and the touch works only when I switch off the sd card task. 
These are the pins I configured in User_Setup.h
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15
#define TFT_DC   2 
#define TFT_RST  4
#define TOUCH_CS 21
#define SD_CS 5

I have even tried to make all the CS pins high initially in setup()
Kindly help me out.

microtronics, 2021-06-24:
Some SD-shields such as this: [MicroSD shield](https://www.amazon.com/gp/product/B077MCQS9P) don't go into high-impedance when releasing CS. In This case you either need to do a hardware-mod (in some cases adding a resistor is enough, though making the level shifter go HiZ is a cleaner solution) or use a different SPI.
In case you are using a XPT2046 touch driver, I found that PaulStoffregen's library doesn't seem to release CS properly. The integrated touch driver doesn't have this issue

HKpro2090, 2021-06-24:
I did not use PaulStoffregen's Library, I used the Bodmer's TFT_espi. Actually, when I use the sd card reader present in the TFT screen and PaulStoffregen' Library for touch it is working properly.

unbrainen, 2021-06-29:
Find the line **#define SUPPORT_TRANSACTION** in User_Setup.h and uncomment it
It helped me on ILI9488

Bodmer, 2021-09-14:
Glad you have a solution. Thanks for posting this.

closed 2021-09-14 by Bodmer