Bodmer/TFT_eSPI #1244: how to get clibration data

rickyanwar, 2021-06-25:
I was using ILI ili9341 (320x240), and esp32 with this example, i was modifying
```
#include "SPIFFS.h"
#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

#define CALIBRATION_FILE "/calibrationData"

void setup(void) {
  uint16_t calibrationData[5];
  uint8_t calDataOK = 0;

  Serial.begin(115200);
  Serial.println("starting");

  tft.init();

  tft.setRotation(0);
  tft.fillScreen((0xFFFF));

  tft.setCursor(20, 0, 2);
  tft.setTextColor(TFT_BLACK, TFT_WHITE);  tft.setTextSize(1);
  tft.println("calibration run");

  // check file system
  if (SPIFFS.begin(true)) {
    Serial.println("formating file system");

    SPIFFS.format();
    SPIFFS.begin();
  }

  // check if calibration file exists
  if (SPIFFS.exists(CALIBRATION_FILE)) {
    File f = SPIFFS.open(CALIBRATION_FILE, "r");
    if (f) {
      if (f.readBytes((char *)calibrationData, 14) == 14)
        calDataOK = 1;
      f.close();
    }
  }
  if (calDataOK) {
    // calibration data valid
    tft.setTouch(calibrationData);
  } else {
    // data not valid. recalibrate
    tft.calibrateTouch(calibrationData, TFT_WHITE, TFT_RED, 15);
    // store data
    File f = SPIFFS.open(CALIBRATION_FILE, "w");
    if (f) {
      f.write((const unsigned char *)calibrationData, 14);
      f.close();
    }
 
    for (int i=0; i < 5; i++){
        Serial.println("");
        Serial.printf("%d", "%d\n", i, calibrationData[i]);  
    }
  }

  tft.fillScreen((0xFFFF));

}

void loop() {
  uint16_t x, y;
  static uint16_t color;

  if (tft.getTouch(&x, &y)) {

    tft.setCursor(5, 5, 2);
    tft.printf("x: %i     ", x);
    tft.setCursor(5, 20, 2);
    tft.printf("y: %i    ", y);

    tft.drawPixel(x, y, color);
    color += 155;
  }
}
```
when I print on serial I got same value
```
 for (int i=0; i < 5; i++){
        Serial.println("");
        Serial.printf("%d", "%d\n", i, calibrationData[i]);  
    }
```

[image]

Pablo2048, 2021-06-25:
The screenshot is definitely not from the code given. `Serial.printf("%d", "%d\n", i, calibrationData[i]);` has to produce different output than '1061162837'...

rickyanwar, 2021-06-25:
how i can i gen calibration data?

Pablo2048, 2021-06-25:
gen or get? (create or read?) Please see my updated comment a correct your code. Because your original formatting `"%d", "%d\n"` is bad.

rickyanwar, 2021-06-25:
> gen or get? (create or read?) Please see my updated comment a correct your code. Because your original formatting `"%d", "%d\n"` is bad.

i following this tutorial

Pablo2048, 2021-06-25:
No, you did not have the same code. See the line with `Serial.printf("%d, %d\n", i, calibrationData[i])` and compare it with yours. YOUR FORMAT STRING IS BAD!!! as I said before.

rickyanwar, 2021-06-25:
ok my mistake, done i get this value

closed 2021-06-25 by rickyanwar