Bodmer/TFT_eSPI #1246: Issues getting TFT_eSPI to compile on ESP32-S2

gregsqueeb, 2021-06-30:
Hello, I am having issues compiling TFT_eSPI on a [Lilygo T8 board](https://github.com/Xinyuan-LilyGO/LilyGo-T-Display-S2). I know that ESP32-S2 boards are not fully supported right now, but I was hoping someone here has been successful in using this library on them and could point me in the right direction. I am kind of at a loss right now.

I have defined my `User_Setup.h` as follows:

```#define VSPI SPI2_HOST

// Setup for the TTGO T4 ("Bitcoin Tracker") ESP32 board with 2.2" ILI9341 display

// See SetupX_Template.h for all options available

#define ST7789_DRIVER

#define TFT_WIDTH 135
#define TFT_HEIGHT 240

#define CGRAM_OFFSET // Library will add offsets required

#define TFT_MOSI 35
#define TFT_SCLK 36
#define TFT_MISO 4
#define TFT_CS 34
#define TFT_DC 37
#define TFT_RST 38

#define TFT_BL 33 // Display backlight control pin

#define TFT_BACKLIGHT_ON HIGH // HIGH or LOW are options

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY 27000000 // Actually sets it to 26.67MHz = 80/3

#define SPI_READ_FREQUENCY 20000000

#define SPI_TOUCH_FREQUENCY 2500000
```

my main.cpp is 
```//file: main.cpp
#include "Arduino.h"
#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup()
{
    Serial.begin(115200);
    // tft.fillScreen(TFT_WHITE);
}

void loop()
{
    Serial.println("loop");
    delay(1000);
}
```

When I try to compile I am getting this error:
```[1/9] Performing build step for 'bootloader'
[1/1] cd /Users/gregmathews/esp/flock/build/bootloader/esp-idf/esptool_py && python /Users/gregmathews/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /Users/gregmathews/esp/flock/build/bootloader/bootloader.bin
Bootloader binary size 0x4cc0 bytes. 0x2340 bytes (46%) free.
[5/7] Linking CXX executable hello-world.elf
FAILED: hello-world.elf 
: && /Users/gregmathews/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ -mlongcalls  CMakeFiles/hello-world.elf.dir/project_elf_src_esp32s2.c.obj -o hello-world.elf  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp32s2/libesp32s2.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/ulp/libulp.a  esp-idf/usb/libusb.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  esp-idf/TFT_eSPI/libTFT_eSPI.a  esp-idf/arduino/libarduino.a  -Wl,--cref -Wl,--Map=/Users/gregmathews/esp/flock/build/hello-world.map  -Wl,--gc-sections  -fno-rtti  -fno-lto  esp-idf/main/libmain.a  esp-idf/arduino/libarduino.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/coap/libcoap.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/touch_element/libtouch_element.a  esp-idf/usb/libusb.a  esp-idf/TFT_eSPI/libTFT_eSPI.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/console/libconsole.a  esp-idf/json/libjson.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp32s2/libesp32s2.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/ulp/libulp.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /Users/gregmathews/esp/esp-idf/components/esp_wifi/lib/esp32s2/libcoexist.a  /Users/gregmathews/esp/esp-idf/components/esp_wifi/lib/esp32s2/libcore.a  /Users/gregmathews/esp/esp-idf/components/esp_wifi/lib/esp32s2/libespnow.a  /Users/gregmathews/esp/esp-idf/components/esp_wifi/lib/esp32s2/libmesh.a  /Users/gregmathews/esp/esp-idf/components/esp_wifi/lib/esp32s2/libnet80211.a  /Users/gregmathews/esp/esp-idf/components/esp_wifi/lib/esp32s2/libpp.a  /Users/gregmathews/esp/esp-idf/components/esp_wifi/lib/esp32s2/libsmartconfig.a  /Users/gregmathews/esp/esp-idf/components/esp_wifi/lib/esp32s2/libwapi.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs [... cut ...] rence to `TFT_eSPI::TFT_eSPI(short, short)'
/Users/gregmathews/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(main.cpp.obj): in function `__static_initialization_and_destruction_0(int, int)':
/Users/gregmathews/esp/flock/build/../main/main.cpp:6: undefined reference to `TFT_eSPI::TFT_eSPI(short, short)'
```

Any tips on how to get this going would be greatly appreciated!

Bodmer, 2021-09-14:
Adding support for the S2 is on my TODO list but I have yet to buy one of those processors and I probably wont have time to look at this for quite a while. Some users have had success using that processor with SPI displays by using the HSPI SPI port but I have not tried this.

closed 2021-09-14 by Bodmer

Bodmer, 2021-11-18:
The library now supports the ESP32 S2, this has been tested with the Arduino ESP32 board package 2.0.1.

A new setup file example has been added:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70_ESP32_S2_ILI9341.h