Bodmer/TFT_eSPI #1252: using ILI9486 with SPI

ayashkant, 2021-07-05:
Hello sir, 
I am using this driver using SPI i have selected the ILI9486 in the user_setup.h and uncommented the esp32 pins with MOSI = 13, MISO = 12, SCK = 14, CS = 27 and DC = 19 but it is not able communicate properly using SPI and I am not able to print anything in the display. Plz help if u have any solution

Bodmer, 2021-09-14:
I have been busy with other priority activities... do you still need support?

nicolasaw, 2021-10-05:
> I have been busy with other priority activities... do you still need support?

Hi, I might have the same issue. I am using ILI9488 SPI and I am confused which pin configurations should I use, any guidance for that?

Bodmer, 2021-11-27:
Assumed resolved.

@nicolasaw please raise a new issue and explain what you need to know.

closed 2021-11-27 by Bodmer