Bodmer/TFT_eSPI #1256: TFT_eSPI, Nano 33 BLE and st7796s display.

IanBUK, 2021-07-08:
Good evening.

Does TFT_eSPI work with the Nano 33 BLE? I am using the Nano with the st7796s display. All works fine using mcufriend and adafruit_gfx, but I want to use TFT_eSPI so I can get anti-aliased fonts and use lvgl.

I’ve created a user setup file, based on the setup 31 st7796s parallel stm one, commenting out the #define stm line. If I don’t disable that line I get a whole load of compile errors because my board is not an stm one. If I do comment that line, I get compile failures in the example because CONSTRUCTOR_INIT_TFT_DATA_BUS is not defined. So, I assume I need an equivalent of the stm or esp32 processor files.

So, does this library work with the Nano 33 BLE and st7796s? What obvious thing am I missing?

Thanks.

TFT pins shown below.

[image]

_Originally posted by @IanBUK in https://github.com/Bodmer/TFT_eSPI/discussions/1248_

IanBUK, 2021-08-15:
Any suggestions? Does/Can this library work on the Nano 33 BLE please?

Bodmer, 2021-09-14:
The Nano 33 would be treated as a generic processor and only basic Arduino I/O functions used (no optimsed code in the library).

I do not have a Nano 33 to test but I would expect it to work albeit very slowly. I suspect it would be so slow however that LVGL would run too slow to be usable.

Bodmer, 2021-09-14:
I will look into the CONSTRUCTOR_INIT_TFT_DATA_BUS error as that may be a bug.

Bodmer, 2021-09-14:
Ah, I see from the picture you posted that it is an 8/16 bit parallel display. Unfortunately parallel displays are not currently supported for generic processors, only ESP32 and STM32F series.

I suspect an SPI display would work OK as this has been tested on other generic processors, but it would be too slow to use with LVGL as explained above. Both parallel and SPI displays really need optimised drivers to get the performance to acceptable levels.

closed 2021-09-14 by Bodmer