Bodmer/TFT_eSPI #1263: Screen (SSD1963) not showing anything when connected to NodeMCU ESP32s

fdrake97, 2021-07-19:
Just bought a 4.3" tft lcd display from BuyDisplay (3.3V). My issue is that although the backlight turns on (after wiring it to pin 18 and setting it to `HIGH` in `setup()`), that's it, nothing happens on the screen. I selected setup 50, wired it using the default pins, and this is the output of the Read_User_Setup example:

`TFT_eSPI ver = 2.3.70
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = Parallel
Display driver = 1963
Display width  = 480
Display height = 800

TFT_CS   = GPIO 33
TFT_DC   = GPIO 15
TFT_RST  = GPIO 32
TFT_WR   = GPIO 4
TFT_RD   = GPIO 2

TFT_D0   = GPIO 12
TFT_D1   = GPIO 13
TFT_D2   = GPIO 26
TFT_D3   = GPIO 25
TFT_D4   = GPIO 17
TFT_D5   = GPIO 16
TFT_D6   = GPIO 27
TFT_D7   = GPIO 14

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled`

I ran the Colour_Test example; the backlight turns on but nothing shows up. I ran the TFT_ReadWrite_Test, and it just keeps giving error (always reading 2695). I am currently using 30 cm wires, and I know shorter ones are better (ordered 10 cm ones) but I feel like the screen should at least display something, however ugly it may look. I double checked all the wiring to ESP32. Everything is as it should be in the documentation except pin 8 is listed as "tearing effect" which I wasn't sure what to do with, so I left it not connected to anything.

Pin documentation from BuyDisplay
[image]

Wiring
[image]
[image]

fdrake97, 2021-07-20:
Issue has been solved. I do not know why. I disconnected all wires and re-wired it just to make sure everything was ok. Interestingly, the read/write test worked, even though I'm pretty sure the wiring was the same. The color sketch did not work. Then, all I did was unplug the usb from my computer and stick it in the wall (phone charger). The screen turned on. color test works perfectly. I figured it wasn't getting enough power from the laptop. I plugged it back into the laptop, and now it works every time. I am 90% happy it works, 10% annoyed I don't know why, and 100% baffled.

fdrake97, 2021-07-21:
After several more hours of testing, it seems there is some strange issue with initializing the display. When trying to load a new sketch, pretty much 100% of the time the screen does not initialize properly and gets stuck on just a black screen. It seems as though after waiting a while, just unplugging and replugging the ESP32's usb does the trick and the display works fine. If no new sketch is loaded, the display will always work. Then, when loading a new sketch, more often than not, the same thing happens. Display doesn't want to do anything, even when reset, unplugging and replugging, restarting Arduino, restarting computer, etc. Then, after a bit, it decides it wants to cooperate and will work upon unplugging and replugging ESP32 to power.

Bodmer, 2021-09-14:
I think the wires are too long and this is causing signal integrity issues. Try adding more GND wires between the display and ESP32.

Also these displays demand a lot of power, so check the power supply is adequate and not collapsing under the load.  Laptop USB power sometimes gets shut off by the laptop power management system and maybe that is why it worke from a USB battery pack.

These things are very hard for me to identify the root cause, generally these intermittent problems are down to poor connections, power and wires being too long.

closed 2021-09-14 by Bodmer

fdrake97, 2021-09-14:
So I never figured out why it happened but I did find a very basic workaround and I'm wondering if this sheds any light (as well as could potentially help someone who reads this). After the screen turns on and doesn't work, as soon as I touch an uncharged capacitor (100µF) against the 3V3 and GND pins of the display, it turns on and everything works perfectly. I can just uncharge the capacitor and do this again whenever the screen does not initialize. Does this help diagnose the issue by any chance? If not, no worries, I just wanted to see if that means there is some obvious issue somewhere.

Bodmer, 2021-09-14:
I expect the uncharged capacitor generates a power transient and causes the display and/or processor to reset/reboot. I suspect the display is getting garbage commands during power up due to floating logic lines and then gets in a state where it will not receive commands.

It takes time for the processor to configure the I/O lines and set the states, during this time electrical disturbances are seen as commands by the display.

Make sure you have the reset line of the display correctly connected to the processor. The library toggles this when it has configured the I/O lines and thus should put the display in a known state for configuration.

Add a 4K7 pullup to the TFT chip slect line so it is high at power up, this should prevent the display getting garbage commands.