Bodmer/TFT_eSPI #1264: Problem in linking

bluexp1, 2021-07-19:
I am trying to compile one of the example for esp8266. The code compiles without any error but the following error occurs from linker:
```
.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\nodemcuv2\firmware.elf section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nodemcuv2\firmware.elf] Error 1 
```
I do not have any clue how to solve it.

bluexp1, 2021-07-23:
The problem solved by defining flash layout via:
`board_build.ldscript = eagle.flash.4m1m.ld`

Bodmer, 2021-09-14:
Thanks for finding and reporting the solution. I would not have been able to help!

closed 2021-09-14 by Bodmer