Bodmer/TFT_eSPI #1265: Support for R61529

dexter323i, 2021-07-21:
If you search Ali for 3.5 inch display with IPS panel, and capacitive touch interface, you will find all the displays are with R61529 driver (not ILI948x). It is the best possible 3.5 IPS CTP display out there, it is sad it looks like TFT_eSPI doesn't support it.
Thank God it only looks like, but it is not true. Actually it does support the R61529! It works perfectly in Parallel mode, after some tweaking:
https://daumemo.com/ips-lcd-esp32-with-espi-library-and-touch-screen-diy-generator-part-9/

https://github.com/daumemo/IPS_LCD_R61529_FT6236_Arduino_eSPI_Test
(this is not my project/repo, I only found it after some googling)

With some cleanup, the new driver could be easily added to TFT_eSPI. 
(It uses #define R61581_DRIVER what is not exactly true, also the R61581_Init.h looks way-way overcomplicated to me, but I am not an expert in this topic.)

ajsb85, 2021-07-25:
Hi, @dexter323i 
Thanks about the researches.
I will try to help on this task.
Any another documentation or reference, please feel free to send as comment of this issue.
Cheers.

ajsb85, 2021-07-25:
The datasheet
[R61529.pdf](https://github.com/ZinggJM/GFX_TFT/files/6873653/R61529.pdf)

dexter323i, 2021-07-25:
Hi @ajsb85, thanks for your attention!
Well, I have no other useful documentations, except daumemo-s repository. I simply used his three files in TFT_Drivers folder:
https://github.com/daumemo/IPS_LCD_R61529_FT6236_Arduino_eSPI_Test/tree/master/lib/TFT_eSPI/TFT_Drivers
For pin mappings, I already had a working breadboard with ILI9481 display, so I used this breadboard with R61529, and migrated my existing pin mappings into the User_Setup.h file. (I used his #define R61581_DRIVER; and #define ESP32_PARALLEL; commands)
The only modifications I had to do on the breadboard are the IM pins. This is what makes R61529 operate in 8bit Parallel mode:
38 - GND 
39 - GND
40 - 3.3V

Some Aliexpress sellers attach a table in the description what points to the differences compared to ILI9488 or ILI9481:
https://www.aliexpress.com/item/32829704426.html
But I see no real differences in Parallel mode. Maybe they mean voltages should be 2.8V max. Well, it works from 3.3V for me. I will post an update when it fails because of the overvoltage, I promise! :)

Bodmer, 2021-09-12:
See #1327

closed 2021-09-12 by Bodmer

m-menon, 2021-09-22:
> If you search Ali for 3.5 inch display with IPS panel, and capacitive touch interface, you will find all the displays are with R61529 driver (not ILI948x). It is the best possible 3.5 IPS CTP [...]

hi @dexter323i , I too got a couple of boards initially thinking that those were using HX8357 but it turned out that the boards have a newer chip which is the R61529. Wondering if you have been able to get TFT_eSPI to work stably on your project? I am starting to see more displays with the R61529. Perhaps it would be a great addition to the board supported by TFT_eSPI. I may not be able to use the parallel mode because I'll be using with an ESP32.

dexter323i, 2021-11-02:
@m-menon yes, follow the links in the comment you quoted.

m-menon, 2021-11-03:
@dexter323i Cool, thanks.

m-menon, 2022-05-21:
I followed the steps described in the above posts down to the last line. I checked the connections thoroughly. The MCU is an ESP32 and the driver IC is a R61529. I can get the display to work but it has this bluish color and doesn't display correctly. 

[image]

This is the diagnostic
```
[code]
TFT_eSPI ver = 2.4.70
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = Parallel
Display driver = 6158
Display width  = 320
Display height = 480

TFT_CS   = GPIO 27
TFT_DC   = GPIO 14
TFT_RST  = GPIO 26
TFT_WR   = GPIO 12
TFT_RD   = GPIO 13

TFT_D0   = GPIO 15
TFT_D1   = GPIO 4
TFT_D2   = GPIO 23
TFT_D3   = GPIO 22
TFT_D4   = GPIO 21
TFT_D5   = GPIO 19
TFT_D6   = GPIO 18
TFT_D7   = GPIO 25

TFT_BL           = GPIO 5
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

[/code]
```
I tried swapping the color order as well
// #define TFT_RGB_ORDER TFT_RGB // Colour order Red-Green-Blue
// #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

I did try different SPI_FREQUENCY as well - both 27000000 and 40000000

probably someone's encountered something similar? Thanks

m-menon, 2022-05-23:
hi, I got it working following the above steps but now there is a somewhat strange issue. The correct colors should be something like this below (from a demo program I found on the web)
[image]

But what I see instead is this as seen in the below picture - any clues on what could be wrong? I used the latest TFT_eSPI library, then used the files from here https://github.com/daumemo/IPS_LCD_R61529_FT6236_Arduino_eSPI_Test/tree/master/lib/TFT_eSPI/TFT_Drivers
and my own pin mappings for the 8 bit parallel connection ( which I thoroughly checked like 10 times ). Thanks, appreciate any pointers.

[image]

TiagoPaulaSilva, 2022-09-10:
> hi, I got it working following the above steps but now there is a somewhat strange issue. The correct colors should be something like this below (from a demo program I found on the web) [image]
> 
> But what I see instead is this as seen in the below picture - any clues on what could be wrong? I used the latest TFT_eSPI library, then used the files from here https://github.com/daumemo/IPS_LCD_R61529_FT6236_Arduino_eSPI_Test/tree/master/lib/TFT_eSPI/TFT_Drivers and my own pin mappings for the 8 bit parallel connection ( which I thoroughly checked like 10 times ). Thanks, appreciate any pointers.
> 
> Here's a short video of the same https://drive.google.com/file/d/18r54_Ixq9eSY7cyo-xmAb_MWC7ULVyX5/view?usp=sharing
> 
> [image]

mm108, did you find any solution? I have similar problem here (I did the same steps as you)

TiagoPaulaSilva, 2022-09-10:
@mm108, did you find any solution? I have similar problem here (I did the same steps as you)

TiagoPaulaSilva, 2022-09-11:
> @mm108, did you find any solution? I have similar problem here (I did the same steps as you)

Solved! I changed the position of resistors that came to 16-bit bus interface as default. Now, it's working properly!

[image]