Bodmer/TFT_eSPI #1267: Issue With Examples on 4" TFT screen (On wemos D1R 32)

ddrmax, 2021-07-23:
Here is the Setup:
ESP32 board: Wemos D1 R32 (Unmodified) (x2)
Screen: 4inch TFT Touch Shield , SKU: 13587 from Waveshare  (x2)
Screen Size: 320x480
The screen have the dip switches moved to use the side pins instead of ICSP (They are in ISCP mode by default)
What the problem is:
White Screen, stayed the same after trying all of the possible fixes from closed issues.
For tests I used and Arduino mega, both screens tested OK
I hacked the Waveshare touch demo for the UNO to run on the ESP 32 board, all is OK but touch is HS (I don't care about touch)
By hacking the code here is what I created as a base to use in TFT_eSPI:
`//Pin definition equivalence between Arduino Uno and Wemos D1 R32
//#define PIN_D0 
//#define PIN_D1 
#define PIN_D2 26 
#define PIN_D3 25
#define PIN_D4 17
#define PIN_D5 16
#define PIN_D6 27
#define PIN_D7 14
#define PIN_D8 12
#define PIN_D9 13
#define PIN_D10 5
#define PIN_D11 23
#define PIN_D12 19
#define PIN_D13 18
#define TFT_CS PIN_D10
#define TFT_DC PIN_D7 // Data Command control pin
#define TFT_RST PIN_D8 // Reset pin (could connect to NodeMCU RST, see next line)
#define LCD_BL PIN_D9
#define TFT_MISO PIN_D12 
#define TFT_MOSI PIN_D11 
#define TFT_SCLK PIN_D13
`
in setup, to activate the backlight:
`      pinMode(LCD_BL, OUTPUT);
        digitalWrite(LCD_BL, HIGH); // TFT screen chip select`

What kind of other things i need to change to make it work as i already tested the 
  #define RPI_DISPLAY_TYPE
  #define ILI9486_DRIVER
in the code with no changes whatsoever?

ddrmax, 2021-07-23:
the problem seems to be from user_setups. it is possible to bypass user setups to add it on a per project basis instead of modifying before each project?

fvanroie, 2021-08-18:
I have the same hardware and use these defines:

```ini
;***************************************************;
; D1 R32 ESP32 with Waveshare ILI9486 Touch Shield  ;
;               - D1-R32 board aka ESPDUINO-32      ;
;               - ili9486 TFT 3.5" or 4"            ;
;               - xpt2046 touch controller          ;
;***************************************************;

    -D TFT_MISO=19
    -D TFT_MOSI=23
    -D TFT_SCLK=18

    -D RPI_DISPLAY_TYPE=1
    -D ILI9486_DRIVER=1

    -D TFT_WIDTH=320
    -D TFT_HEIGHT=480
    -D TFT_ROTATION=0 ; Use default, see TFT_ROTATION values
    -D SPI_FREQUENCY=20000000 ; Some displays will operate at higher frequencies
    -D SPI_TOUCH_FREQUENCY=2500000
    -D USER_SETUP_LOADED=1  ; Skip User_Setup.h
    -D TOUCH_DRIVER=2046 ; XPT2046
    ;-D SUPPORT_TRANSACTIONS   ; Default on ESP32

    -D TFT_DC=14
    -D TFT_CS=5 
    -D TFT_RST=12
    -D TFT_BCKL=13
    -D TOUCH_CS=17
```

Bodmer, 2021-09-14:
Closing as assumed to be resolved.

@fvanroie Thanks for the support.

closed 2021-09-14 by Bodmer