Bodmer/TFT_eSPI #1270: Multiple ILI9431 displays each with sprites on ESP32

englishacoustics, 2021-07-24:
Dear All,

We are trying to get multiple TFT-ili9431 displays working on an ESP32.  Currently using the multiple GPIO to the CS pins on each display to select the display to update.

This works fine, except the gauge sprites are used (or any sprites for that matter) at which point the displays corrupt and become unreadable where the sprite part would have moved or animated.

We have been trying to resolve this for several days with no luck, we found another post about using dual and triple tfts on github and that works - but not with sprites.

Can anyone offer any advise or help.  I'll post code later, but the question is using the CS pin and sprites doesn't work with multiple displays, all other functions do.

Pete

Bodmer, 2021-08-23:
Use tft.endWrite(); before swapping the display and that will make sure the SPI output buffers are empty before you change the chip select.

Let me know if this works for you.

englishacoustics, 2021-08-24:
> Bug from
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> This message has been scanned for viruses and
> dangerous content by the *Togethia MailScanner*
> and is believed to be clean.
> Togethia logo

Hi Bodmer,

Thankyou very much for this, I'll test it and let you know,

Reards

Pete#

	
Peter Farrow BEng(hons) BBC ETSI
Chief Technical Officer
Email: ***@***.***
***@***.***>
Website: www.englishacoustics.co.uk <https://www.englishacoustics.co.uk>

	+44 (0)7799605617 <tel:+44 (0)7799605617> 		+44(0)1249 736180
<tel:+44(0)1249 654183> 	<https://facebook.com/englishacoustics>
<skype:peter_farrow>

closed 2021-11-21 by Bodmer