Bodmer/TFT_eSPI #1271: Can't make Animated_Eyes example work with Raspberry Pi Pico and GC9A01 Round Display

djairjr, 2021-07-28:
First of all, thank you so much for your dedicated work. I'm trying to make it possible to apply the GC9A01 display for use with the Raspberry Pi Pico. Recently I used the Animated_Eyes example, connecting a Teensy 3.2 to two such displays, and with the correct changes in User_Setup.h it was possible to make it work.

I'm trying to do the same, this time using the Raspberry Pi Pico as controller. But I can't make that example specifically work. I don't get any images on the display. I believe there must be some error in the call to the display, but I could not perform the debug yet. The Raspberry Pi Pico disables debug via USB and has not yet prepared a system to perform debug through the DEBUG port. As soon as I do that, I'll post here the messages I'm getting. On the other hand, the COLOUR_TEST is working perfectly, with both displays simultaneously.

The Teensy User Setup changes is as follows:
`#define GC9A01_DRIVER // Round TFT SPI Display 

#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // GC9A01 240 x 240

// Define TFT Pins to Teensy
#define TFT_MISO 12
#define TFT_MOSI 11
#define TFT_SCLK 13
#define TFT_CS   10  // Chip select control pin
#define TFT_DC   7  // Data Command control pin
#define TFT_RST  8  // Reset pin (could connect to RST pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000`

and the Raspberry Pi Pico is that:

`#define GC9A01_DRIVER // Round TFT SPI Display 

#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // GC9A01 240 x 240

#define TFT_MISO 0 // RX PIN - Not Connected
#define TFT_SCLK 2
#define TFT_MOSI 3 // TX PIN
#define TFT_CS   20  // Chip select control pin
#define TFT_DC   18  // Data Command control pin
#define TFT_RST  19  // Reset pin (could connect to Arduino RESET pin)

#define TFT_BL   17            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  70000000
//#define SPI_FREQUENCY  40000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  5000000`

I hope this can help.

Bodmer, 2021-08-23:
The sketch works on a Raspberry Pico for me.

I note you have set the SPI clock rate to 70MHz:
```
#define SPI_FREQUENCY  70000000
```
 these displays do NOT work at that speed. Try 27MHz (as you have used for the Teensy):
```
#define SPI_FREQUENCY  27000000
```

djairjr, 2021-08-31:
Thank you. I will try it!

djairjr, 2021-08-31:
Confirm working! Just a few changes in original code, to get both eyes centered.
In file eye_functions, change the line:
`tft.setAddrWindow(eye[e].xposition, 0, 128, 128); `
change it to:
`tft.setAddrWindow(eye[e].xposition, 0, 128, 128); `

and in config.h file, change this lines:
`#define TFT_1_ROT 0        // TFT 1 rotation
#define TFT_2_ROT 0        // TFT 2 rotation
#define EYE_1_XPOSITION  60 // x shift for eye 1 image on display
#define EYE_2_XPOSITION  60 // x shift for eye 2 image on display`

djairjr, 2021-09-03:
I purchased a [XIAO RP2040 board](https://wiki.seeedstudio.com/XIAO-RP2040/), which should arrive in early October, so effective testing should only happen around that date. But looking at the documentation, I created this USER_SETUP to use that board with the GC9A01 display.

`
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

#define GC9A01_DRIVER // Round TFT SPI Display 

#define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 240 // GC9A01 240 x 240

#define TFT_MISO 9 // RX PIN - Not Connected
#define TFT_SCLK 8
#define TFT_MOSI 10 // TX PIN
#define TFT_CS   7  // Chip select control pin
#define TFT_DC   4  // Data Command control pin
#define TFT_RST  5  // Reset pin (could connect to Arduino RESET pin)

#define TFT_BL   6            // LED back-light control pin
#define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
#define PIXEL_DOUBLE // Testando

#define SPI_FREQUENCY  27000000
//#define SPI_FREQUENCY  40000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  5000000
// #define SPI_TOUCH_FREQUENCY  2500000 - Copy from Setup200_GC9A01
`

When I have taken the test, I publish the conclusions here. 
Thank you one more time.

Bodmer, 2021-09-14:
Glad you have it working.

Raise another issue if you have problems with the new board.

closed 2021-09-14 by Bodmer

Vince579, 2024-09-16:
Hi,

I have two 0.71 inch IPS displays and also the needed GC9D01 driver. But i can´t make it work with the raspberry pi pico.
It is the first time i work with these stuff. I don´t have really an idea what i have to change. Can someone help me?

[GC9D01.zip](https://github.com/user-attachments/files/17010503/GC9D01.zip)

These are the displays:
https://spotpear.com/shop/Raspberry-Pi-ESP32-Pico-Arduino-STM32-51-0.71-Round-LCD-EYE.html

Best regards
Vince