Bodmer/TFT_eSPI #1274: Unexpected pushSprite behavior

ghdi, 2021-07-31:
Hi,

First of all, I really like your library, great work.

Secondly, I believe I have bumped into an issue with the following method:

`bool     pushSprite(int32_t tx, int32_t ty, int32_t sx, int32_t sy, int32_t sw, int32_t sh);`

It only seems to work for me when I push the full sprite. The moment I start cropping the sprite, the text no longer displays. See code below. Did I misunderstand the working of this method or is this a bug?

```

#include <TFT_eSPI.h>

const char *text = "A STRING THAT WILL DEFINITELY BE LONGER THAN MY 240PX SCREEN USING FONT 4";

TFT_eSPI tft = TFT_eSPI();

TFT_eSprite s = TFT_eSprite(&tft);

void setup() {
  
  tft.init();
  tft.fillScreen(TFT_BLACK);
  tft.setRotation(2);

  s.setColorDepth(1);
  s.createSprite(240, tft.fontHeight(4));
  s.fillSprite(TFT_BLACK);
  s.setTextColor(TFT_WHITE);
  s.setTextDatum(TL_DATUM);
  tft.setBitmapColor(TFT_ORANGE, TFT_BLACK);

  s.drawString(text, 0, 0, 4);

  // This works
  s.pushSprite(0,0, 0, 0, 240, tft.fontHeight(4));

  // This doesn't work, but why? I'd expect a cropped sprite 239px wide to be pushed at screen coordinates 0.0
  //s.pushSprite(0,0, 0, 0, 239, tft.fontHeight(4));

  // This doesn't work, but why? I'd expect a cropped sprite 238px wide to be pushed at screen coordinates 0.0
  //s.pushSprite(0,0, 1, 0, 238, tft.fontHeight(4));
}

void loop()
{
}
```

Bodmer, 2021-08-23:
Thanks for reporting this.

Yes, you have found a bug, this only affects 1 bpp sprites. I will investigate.

Bodmer, 2021-12-11:
This has been fixed, example code above operates as expected.

closed 2021-12-11 by Bodmer