Bodmer/TFT_eSPI #1278: Raspberry Pi Pico - Waveshare ST7789 Pico-LCD not working

zpm1066, 2021-08-05:
I'm unable to get `TFT_eSPI` working correctly with a `Waveshare Pico-LCD HAT (ST7789)` on a `Raspberry Pi Pico`. 

The environment and display connections are: 
```
Dev. Environment: earlphilhower arduino-pico

LCD -> Pico
LCD_DC - TFT_DC, GP8
LCD_CS - TFT_CS, GP9
LCD_CLK - TFT_SCLK, GP10
LCD_DIN - TFT_MOSI, GP11 
LCD_RST - TFT_RST, GP12
LCD_BL - GP13
```
I've tried changing the `TFT_MOSI & TFT_SCLK` pins to the `Waveshare Pico-LCD HAT` using `SPI.setSCK() & SPI.setTX()` but I get a blank display. 

On a standalone Waveshare ST7789 display, TFT_eSPI works fine with the following connections.
```
LCD_DC - TFT_CS, GP8
LCD_CS - TFT_CS, GP9
LCD_CLK - TFT_SCLK, GP2
LCD_DIN - TFT_MOSI, GP3 
LCD_RST - TFT_RST, GP12
LCD_BL   - GP13
```

Is there a way I can get `TFT_eSPI` to work with the `Waveshare Pico-LCD HAT`? Thank you.

zpm1066, 2021-08-06:
It looks like the TFT_eSPI defaults to using SPI0, and the Waveshare Pico-LCD HAT uses SPI1.

How can I modify TFT_eSPI to use SPI1? Thanks.

Bodmer, 2021-08-23:
Changling spi0 to spi1 in the TFT_eSPI.cpp and Processors/TFT_eSPI_RP2040.h and Processors/TFT_eSPI_RP2040.c should work.

LEt me know how you get on and it can be made a configuration option (as is used for the STM32 processors).

zpm1066, 2021-08-23:
Before posting the issue, I had tried replacing all occurrences of `spi0` with `spi1` in the the `TFT_eSPI.cpp and Processors/TFT_eSPI_RP2040.h and Processors/TFT_eSPI_RP2040.c`. However, the switch to `spi1` didn't work.

Anyway, I have just tried changing `spi0` to `spi1` again. I got the same result. Nothing on the display.

I used the following pinouts:

```
Waveshare Pico-LCD HAT -> Pico
LCD_DC - TFT_DC, GP8
LCD_CS - TFT_CS, GP9
LCD_CLK - TFT_SCLK, GP10
LCD_DIN - TFT_MOSI, GP11 
LCD_RST - TFT_RST, GP12
LCD_BL - GP13
```

Bodmer, 2021-08-23:
I don't have the time to look into this unfortunately.
One kludge is tou use spi0 on pins 4,6,7 and wire them across to the display pins. See [RP2040 data sheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf) section 1.4.3 for pins that can use spi0.

zpm1066, 2021-08-24:
OK, understood that it's more involved than a simple switch from spi0 to spi1.

Since posting the issue, I've been using spi0 with GP2 (SCK) & GP3 (MOSI) wired across to GP10 (SPI1 SCK) & GP11 (SPI1 MOSI). This works fine, as I won't be using SPI0 & SPI1 at the same time. I was just hoping for a simple switch over from SPI0 to SPI1.

The Waveshare displays for the RP2040 have their pins wired for SPI1, same as Adafruit ST7789 display library.

skerr92, 2021-09-03:
I've been fighting to get the Adafruit ST7789 breakout to work with the RP2040. I've attempted to use the following pin configuration:

```cpp
#define TFT_SCLK 2 // GPIO 2
#define TFT_MOSI 3 // TX PIN // GPIO 3
#define TFT_CS   17  // Chip select control pin GPIO 20
#define TFT_DC   20  // Data Command control pin // GPIO 18
#define TFT_RST  19  // Reset pin (could connect to Arduino RESET pin) GPIO 19
```

I tried to change ```spi0``` to ```spi1``` but I get a compiler error. Has anyone made progress with this?

zpm1066, 2021-09-11:
Bodmer has marked this issue as a “To do: enhancement”, so RP2040 SPI1 support may happen in the future. 

For now, as Adafruit use GP10/GP11 for SCLK/MOSI, you can just wires from GP2/GP3 to GP10/GP11. The display should then work fine.

skerr92, 2021-09-11:
Turns out the issue I was having came down to using the RP2040 Arduino core vs the official MBed OS RP2040 core. If you try to use the Official Arduino MBed OS RP2040 board manager, there is an error the causes the code not to work. If you use this [RP2040 Arduino package](https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json) it works. Not sure why that ends up being the case. Hope this helps with development.

Bodmer, 2021-09-12:
Thanks for reporting this. The two different Arduino RP2040 support packages appear to allocate different default SPI pins for the board variations. The board suppliers also seem to allocate different pins which adds a further level of variation.

Glad to hear you have it working now. I will add a TODO for making the SPI allocations configurable.

closed 2021-09-12 by Bodmer