Bodmer/TFT_eSPI #1282: I have a 2.4 inch ILI9341 8 bit with XPT2046 and it gives me an error when I try to use spi touch

hannescam, 2021-08-09:
error message:

`Arduino: 1.8.13 (Linux), Board: "ESP32 Dev Module, Disabled, Huge APP (3MB No OTA/1MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"
In file included from /home/hannes/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:4973:0:
/home/hannes/Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp: In member function 'void TFT_eSPI::begin_touch_read_write()':
/home/hannes/Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp:23:34: error: 'spi' was not declared in this scope
     if (locked) {locked = false; spi.beginTransaction(SPISettings(SPI_TOUCH_FREQUENCY, MSBFIRST, SPI_MODE0));}
                                  ^
/home/hannes/Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp: In member function 'void TFT_eSPI::end_touch_read_write()':
/home/hannes/Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp:38:54: error: 'spi' was not declared in this scope
     if(!inTransaction) {if (!locked) {locked = true; spi.endTransaction();}}
                                                      ^
/home/hannes/Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp: In member function 'uint8_t TFT_eSPI::getTouchRaw(uint16_t*, uint16_t*)':
/home/hannes/Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp:62:3: error: 'spi' was not declared in this scope
   spi.transfer(0xd0);                    // Start new YP conversion
   ^
/home/hannes/Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp: In member function 'uint16_t TFT_eSPI::getTouchRawZ()':
/home/hannes/Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp:105:3: error: 'spi' was not declared in this scope
   spi.transfer(0xb0);               // Start new Z1 conversion
   ^
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`

Bodmer, 2021-08-23:
If you use a parallel display then you cannot use the touch support added to the library.

Post the output from the "Read_User_Setup" sketch so I can see your settings.

Alfred586, 2021-08-29:
@hannescam 
You should use an external xpt2046 library.

hannescam, 2021-08-29:
i use an external library but that is for me not a solution i cant simply make a button with 1 command

Bodmer, 2021-09-14:
Unfortunately there is no built in support within TFT_eSPI for parallel screens with resistance based touch screens. The other issue is that the ESP32 runs out of analogue pins so csome pins must be used with dual functionality. See [this section](https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-support) and the[ link](https://github.com/s60sc/Adafruit_TouchScreen) at the bottom of that section.

closed 2021-09-14 by Bodmer

hannescam, 2021-09-15:
thankyou i will try that but what external library i shuold use and what code to create a button and evreything else?