Bodmer/TFT_eSPI #1291: error: a15 cannot be used in asm here

infrafast, 2021-08-17:
Hello I recently switched from ArduinoIDE 1.8 environement to PlatformIO
Everything compiles fine on both environnement with the following exception:

on platformIO, if you add below option in the platformIO.ini file
build_type = debug

(this flag is used to benefit from the exception decoder line information)

when you compile, you get the following error:

lib\TFT_eSPI\TFT_eSPI.cpp:1338:1: error: a15 cannot be used in asm here
 }
 ^
*** [.pio\build\debug_version\libf13\TFT_eSPI\TFT_eSPI.cpp.o] Error 1

This happens with version [2.7.4 ](https://github.com/esp8266/Arduino/releases/tag/2.7.4)of the Arduino core
using Mini D1 pro chip.

No clue if this can be easily fix or not... but reporting it in case it may help. 
Thank you

Bodmer, 2021-08-23:
Unfortunately I am unable to help. Maybe someone on the PlatformIO forum can advise, if so then please post back the answer.

closed 2021-08-23 by Bodmer

infrafast, 2022-03-11:
This can be fixed by adding

-D PGM_READ_UNALIGNED=0

in the platformIO.ini file