Bodmer/TFT_eSPI #1292: how i create one new class that inherited tft_eSpi?

alieslamibidgoli, 2021-08-18:
I wrote this code but Compiler gets an error 

public class Box : tft_eSprite{
}

Pablo2048, 2021-08-18:
https://www.geeksforgeeks.org/inheritance-in-c/

```cpp
class Box : public TFT_eSprite {
};
```

same way as in https://github.com/Bodmer/TFT_eSPI/blob/58f457ba9703613bfcd70b411d66bbbc948fa815/Extensions/Sprite.h#L8

alieslamibidgoli, 2021-08-18:
> 
> 
> https://www.geeksforgeeks.org/inheritance-in-c/
> 
> ```c++
> class Box : public TFT_eSprite {
> };
> ```
> 
> same way as in
> 
> https://github.com/Bodmer/TFT_eSPI/blob/58f457ba9703613bfcd70b411d66bbbc948fa815/Extensions/Sprite.h#L8

thank you.
but "class Box : public TFT_eSprite {};" gets error.

Pablo2048, 2021-08-19:
... and the error is?

Bodmer, 2021-08-23:
If you have spotted an issue with the library I can help, but it is not clear to me what issue you have with the library.

closed 2021-08-23 by Bodmer