Bodmer/TFT_eSPI #1294: Support for ESP32 S2 for RPi ili 9486 3.5" display

BindasJP, 2021-08-19:
How can ESP32 S2 be connected to RPi display with ILI9486 display ? With ESP32 it works fine but with ESP32 S2 even after changing SPI defines to FSPI, it is not working. Any directions ?

Bodmer, 2021-08-23:
The ESP32 S2 is not currently supported though some users have had some success.

BindasJP, 2021-08-24:
Thanks for your reply. I did try and for that matter 240 * 240 display with ST 7789 chip is working fine with SPI and with ILI9486 also touch is working fine so I think only some minor tweaking is needed somewhere for this . Well, I will try but if possible , please do have a look into this issue. Thanks.

sbonaime, 2021-09-01:
@BindasJP Can you explain us how you manage to use an ST 7789 with esp32-s2 ? Thanks

closed 2021-11-18 by Bodmer