Bodmer/TFT_eSPI #1299: Parallel support for RP2040

DustinWatts, 2021-08-20:
Hi Bodmer!

Do you have any plans to support 8/16 bit parallel in your awesome library? There is a demo here using the PIO: https://github.com/zapta/pio_tft/tree/main/platformio/src

Would be very interesting to have RP2040 PIO support in TFT_eSPI!

Cheers!

Bodmer, 2021-08-23:
Adding parallel support is on the TODO list and using PIO would be great.  Thanks for the link that makes a good starting point.

I just need more free time to implement it!

DustinWatts, 2021-08-24:
Where can I donate free time? 😄

Bodmer, 2021-12-16:
There is a new branch here:

https://github.com/Bodmer/TFT_eSPI/tree/RP2040_8bit_parallel

This implements 8 bit parallel support for the RP2040 using the PIO hardware to manage the interface. This provides a high data transfer rate to the screen (10ms to clear 480x320 pixel screen).

Important: At the moment it ONLY works with the Earle Philhower RP2040 board package.

Bodmer, 2022-01-02:
RP2040 parallel support is now built into the main branch, this uses the PIO capability in the RP2040 processor. The PIO is also used for block fills so, for example, the processor can carry on executing software while the PIO fills the screen with a colour. DMA is also supported at >30M pixels/s.

Important: At the moment it ONLY works with the Earle Philhower RP2040 board package.

closed 2022-02-13 by Bodmer

brightproject, 2023-04-22:
Excuse me, how do I connect an 8 or 16 bit display to RP2040-zero?
Is this possible and what wiring diagram should I use?
[image]

Bodmer, 2023-04-22:
Example connections for the GPIO:
```
#define TFT_RST   8 // Reset pin
#define TFT_CS    9  // Do not define if chip select control pin permanently connected to 0V
//#define TFT_RD   -1   // Do not define, read pin must be permanently connected to 3V3

// Note: All the following pins are PIO hardware configured and driven
  #define TFT_WR   10    // Write strobe pin
  #define TFT_DC   11    // Data Command control pin

  // PIO requires these to be sequentially increasing
  #define TFT_D0    0
  #define TFT_D1    1
  #define TFT_D2    2
  #define TFT_D3    3
  #define TFT_D4    4
  #define TFT_D5    5
  #define TFT_D6    6
  #define TFT_D7    7
```
You will also need to connect up power and maybe a backlight control signal depending on the display you have.