Bodmer/TFT_eSPI #1300: Problem readRect with R61581

LvlLord, 2021-08-20:
Hello,
I had problems regarding readRect with the R61581 display. I used the "Animated_dial" test and everywhere where the dial was, the image got distorted.
It seems that the R61581 also returns three bytes (the 24 bit RGB value part), because when I hardcoded this variant, the effect was gone.
Has anybody else this device to test and confirm?

Hope this is the right section to put issues with a solution :)

Markus

Bodmer, 2021-08-23:
I do not have an R61581 display, so am unable to advise.

The changes to add the R61581 were added by a user ([see here](https://github.com/Bodmer/TFT_eSPI/issues/238)) so I suspect the reading from the TFT was left out.

Fischkopppp, 2022-03-10:
I have this display. I can try it over the weekend.

Bodmer, 2022-03-12:
@Fischkopppp: Is that a SPI or 8 bit parallel UNO style display you have?

Fischkopppp, 2022-03-12:
It is a 8bit parallel display. I am using this library: https://github.com/daumemo/IPS_LCD_R61529_FT6236_Arduino_eSPI_Test based on an older TFT_eSPI version. It is the same display as described here: https://daumemo.com/ips-lcd-esp32-with-espi-library-and-touch-screen-diy-generator-part-9/

Fischkopppp, 2022-03-12:
@LvlLord  could you send me you UserSetup or Platformio.ini settings?

Bodmer, 2022-11-05:
No response from @LvlLord so closing.

closed 2022-11-05 by Bodmer