Bodmer/TFT_eSPI #1301: ESP32 SPI DMA not enabled when using TFT_eSPI with Arduino as a component in ESP-IDF

caiser01, 2021-08-20:
Today, I encountered an issue with slow screen refreshes when using TFT_eSPI with Arduino as a component under ESP-IDF. I was able to track the problem down to ESP32_DMA not being enabled.

When I use the TFT_eSPI library in the Arduino IDE or in PlatformIO with just the Arduino framework, ESP32_DMA is enabled seemingly by default and screen refreshes are quick as expected. If I instead use ESP-IDF with Arduino as a component in PlatformIO, ESP32_DMA is not enabled by default and screen refresh performance suffers.

Here are two simple test programs to illustrate. More configuration details are in the platform.ini for each project. I've tested this on both ESP32-WROOM-32D and ESP32-WROVER-E modules and the results are the same.

This one uses just Arduino framework: https://github.com/caiser01/Arduino_TFT_DMA_check

This one uses ESP-IDF with Arduino as a component: https://github.com/caiser01/ESP-IDF_TFT_DMA_check

Both programs check whether ESP32_DMA is enabled and then measure the time it takes to fill the screen with solid red, green, and blue. Here is sample output I got from running each program:

#### PlatformIO, Arduino framework:
`[D][main.cpp:9] setup(): ESP32 DMA enabled`
`Hello from Arduino!`
`Draw time was: 371805 microseconds.`

#### PlatformIO, ESP-IDF with Arduino as component:
`[D][main.cpp:50] setup(): ESP32 DMA not enabled`
`Hello from Arduino!`
`Draw time was: 1389782 microseconds.`

As you can see the draw time in the second case is more than 3 times greater than in the first.

My question is, how do I get ESP32_DMA to be enabled for the second case where I'm using ESP-IDF with Arduino as a component?

Thanks!

Bodmer, 2021-08-23:
In the Arduino environment "ESP32" is defined so "ESP32"_DMA" is then automatically defined by the library.

You can add the definition in the "platformio.ini" file, e.g.:

```
  -DESP32_DMA=1
```

closed 2021-08-23 by Bodmer

caiser01, 2021-08-23:
Adding just:

`-DESP32=1`

to platform.ini was enough to resolve it.

Thanks for your help and for creating and maintaining this great library!