Bodmer/TFT_eSPI #1302: Issues getting examples to build in platformio, for RP2040 target with ILI9341 driver

MakerMatrix, 2021-08-22:
It seems there are many prototypes missing.  Is this a thing that the Arduino IDE lets you get away with but not a more standard compiler?

Also I get hundreds of pages of warnings about macros (in particular whenever spi0, spi0_*, SPI0_*, etc) get expanded, but many others too), and various warnings about variable length arrays.

None of the warnings prevent the code from compiling (the missing prototypes do but it is easily but tediously fixed).  I don't believe the warnings are your issue anyway - If I understand what is going on, you are calling the macro which gets expanded out of framework-arduino-mbed and that code generates the various warnings.  It's just incredibly annoying and worrysome, and I'm wondering if you have advice on how to fix it or what I need to ask the upstream RP2040 people to do about it?

Here's part of a screen scrape to give you a flavor:

```
C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_spi/include/hardware/spi.h:49:35: note: in expansion of macro 'spi0_hw'
   49 | #define spi0 ((spi_inst_t * const)spi0_hw)
      |                                   ^~~~~~~
.pio\libdeps\pico\TFT_eSPI\Processors/TFT_eSPI_RP2040.h:192:78: note: in expansion of macro 'spi0'
  192 |     #define tft_Write_16(C)     while (!spi_is_writable(spi0)){}; spi_get_hw(spi0)->dr = (uint32_t)(C)
      |                                                                              ^~~~
.pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp:4218:18: note: in expansion of macro 'tft_Write_16'
 4218 |         if (pX) {tft_Write_16(textbgcolor);}
      |                  ^~~~~~~~~~~~
In file included from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/platform_defs.h:12,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/pico_platform/include/pico/platform.h:12,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_base/include/pico.h:19,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/objects.h:30,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/device.h:35,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/platform/include/platform/platform.h:28,       
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/platform/include/platform/FileHandle.h:25,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/macros.h:41,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO/pins_arduino.h:2,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/Arduino.h:76,
                 from .pio\libdeps\pico\TFT_eSPI\TFT_eSPI.h:30,
                 from .pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp:17:
C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:50:19: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   50 | #define SPI0_BASE 0x4003c000
      |                   ^~~~~~~~~~
C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/spi.h:26:35: note: in expansion of macro 'SPI0_BASE'
   26 | #define spi0_hw ((spi_hw_t *const)SPI0_BASE)
      |                                   ^~~~~~~~~
C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_spi/include/hardware/spi.h:49:35: note: in expansion of macro 'spi0_hw'
   49 | #define spi0 ((spi_inst_t * const)spi0_hw)
      |                                   ^~~~~~~
.pio\libdeps\pico\TFT_eSPI\Processors/TFT_eSPI_RP2040.h:192:57: note: in expansion of macro 'spi0'
  192 |     #define tft_Write_16(C)     while (!spi_is_writable(spi0)){}; spi_get_hw(spi0)->dr = (uint32_t)(C)
      |                                                         ^~~~
.pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp:4267:30: note: in expansion of macro 'tft_Write_16'
 4267 |               while (tnp--) {tft_Write_16(textcolor);}
      |                              ^~~~~~~~~~~~
In file included from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_spi/include/hardware/spi.h:12,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/objects.h:41,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/device.h:35,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/platform/include/platform/platform.h:28,       
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/platform/include/platform/FileHandle.h:25,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/macros.h:41,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO/pins_arduino.h:2,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/Arduino.h:76,
                 from .pio\libdeps\pico\TFT_eSPI\TFT_eSPI.h:30,
                 from .pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp:17:
C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/spi.h:26:44: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   26 | #define spi0_hw ((spi_hw_t *const)SPI0_BASE)
      |                                            ^
C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_spi/include/hardware/spi.h:49:35: note: in expansion of macro 'spi0_hw'
   49 | #define spi0 ((spi_inst_t * const)spi0_hw)
      |                                   ^~~~~~~
.pio\libdeps\pico\TFT_eSPI\Processors/TFT_eSPI_RP2040.h:192:57: note: in expansion of macro 'spi0'
  192 |     #define tft_Write_16(C)     while (!spi_is_writable(spi0)){}; spi_get_hw(spi0)->dr = (uint32_t)(C)
      |                                                         ^~~~
.pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp:4267:30: note: in expansion of macro 'tft_Write_16'
 4267 |               while (tnp--) {tft_Write_16(textcolor);}
      |                              ^~~~~~~~~~~~
In file included from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/platform_defs.h:12,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/pico_platform/include/pico/platform.h:12,
                 from C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_base/include/pico.h:19,
[... 53 lines cut ...]
      |        ^~~~~~
.pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp: In member function 'int16_t TFT_eSPI::drawString(const arduino::String&, int32_t, int32_t, uint8_t)':      
.pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp:4366:8: warning: variable length array 'buffer' is used [-Wvla]
 4366 |   char buffer[len];
      |        ^~~~~~
.pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp: In member function 'int16_t TFT_eSPI::drawCentreString(const arduino::String&, int32_t, int32_t, uint8_t)':.pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp:4611:8: warning: variable length array 'buffer'  [... cut ...]
.pio\libdeps\pico\TFT_eSPI\Extensions/Sprite.cpp:485:12: warning: variable length array 'sline_buffer' is used [-Wvla]
  485 |   uint16_t sline_buffer[max_x - min_x + 1];
      |            ^~~~~~~~~~~~
.pio\libdeps\pico\TFT_eSPI\Extensions/Sprite.cpp: In member function 'bool TFT_eSprite::pushToSprite(TFT_eSprite*, int32_t, int32_t, uint16_t)':
.pio\libdeps\pico\TFT_eSPI\Extensions/Sprite.cpp:764:12: warning: variable length array 'sline_buffer' is used [-Wvla]
  764 |   uint16_t sline_buffer[width()];
```

Bodmer, 2021-08-23:
It looks like the Arduino support structs are not being loaded.

Taking the first undefined _spi0_hw:_
```
#define spi0 ((spi_inst_t * const)spi0_hw)
```
This line is found in the header here. ...pico-sdk\src\rp2_common\hardware_spi\include\hardware\spi.h

The required definitions are in the header here ...pico-sdk\src\rp2040\hardware_structs\include\hardware\structs\spi.h

The "variable length array" will be updated at some point but this is not a priority at the moment.

closed 2021-08-23 by Bodmer

MakerMatrix, 2021-11-13:
Finally got back to this project...

#include <spi.h>

instead of:

#include "spi.h"

Fixed the VAST majority of the warnings.  However now I'm left with this:

C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:62:18: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   62 | #define DMA_BASE 0x50000000
      |                  ^~~~~~~~~~
C:\Users\jsmith\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/dma.h:61:34: note: in expansion of macro 'DMA_BASE'
   61 | #define dma_hw ((dma_hw_t *const)DMA_BASE)

It looks like you are casting DMA_BASE as a const but it's a macro.  Not sure if you care or not.  I suppose it's a somewhat pedantic compiler warning that I could turn off...but for completeness I wanted to report it.

Thanks, I love your libraries and your work on the RP2040 is just awesome.

Bodmer, 2021-11-17:
The reported warnings are part of the sdk and not the TFT_eSPI library so need to be reported in the appropriate repository. You can see the problem file location by examining the file paths.

Glad to hear you are enjoying using the library, thanks for the feedback.