Bodmer/TFT_eSPI #1305: ST7796 readPixel issue

DanielAndrascik, 2021-08-23:
Hello

Reading color from ST7796 over spi returns 16bit color code (565 format). 
But according to the attached image is there one dumy cycle and needs to be read 17bits.
[image]

Bodmer, 2021-08-23:
This should be handled by a bit shift this is in the code. Are you saying the wrong colours are being returned?

DanielAndrascik, 2021-08-24:
> 
> 
> This should be handled by a bit shift this is in the code. Are you saying the wrong colours are being returned?

Yes. I was try this:

tft.fillScreen(TFT_BLUE); //0x1F -> 2b0000 0000 0001 1111
uint16_t c = tft.readPixel(1, 1);

In c is: 0x0F -> 2b0000 0000 0000 1111

One bit is missing.

I tried modify readPixel metod and i replaced this code:

    #if defined (ST7796_DRIVER)
      // Read the 2 bytes
      color = ((tft_Read_8()) << 8) | (tft_Read_8());
    #else

with this code:

    #if defined (ST7796_DRIVER)
      uint32_t r = spi.transfer32(0); // this return 2b0000 0000 0000 1111 1000 0000 0000 0000
      color = r >> 15;
    #else

Bodmer, 2021-12-12:
I have run some tests and can only reproduce the problem if a high SPI read clock frequency of 40MHz is used.

I have checked and there is no extra clock pulse for colour 16 bit CGRAM reads. The data hints at this and the extra pulse is only for certain register reads, eg Read Display Status (RDDST).

The read clock frequency should not be higher than 16MHz. The test frequencies I used are:
```
#define SPI_FREQUENCY       40000000
#define SPI_READ_FREQUENCY  16000000
```
What I suspect is happening is that the TFT is slow to put out the data bits, so the next clock cycle catches the data from the previous clock slot. I confirmed this by increasing the frequency of the read clock, the data appears to shift right, losing the last bit.

In the data sheet the minimum SPI clock cycle time is 150ns, which is 6.67MHz maximum however mine reads pixel data fine at 16MHz but not 20MHz.

In conclusion I think reading data at high clock rates with a bit shift is causing bit smearing into the next clock period so is likely to be unreliable.

I recommend the read frequency is therefor reduced until bit shifts are not needed. There is a "walking 1" read pixel test sketch you can use to check data validity.

closed 2021-12-12 by Bodmer

Bodmer, 2021-12-12:
Update. My ST7796 display is marginal at 16MHz, so I have had to reduce the SPI clock to 13.4 MHz to avoid pixel errors for large block reads. This is still over-clocking the read bit rate by a factor of more than 2 on the maximum data sheet figures.