Bodmer/TFT_eSPI #1311: Addition of STM_PORTC_DATA_BUS

toogooda, 2021-08-28:
Hi firstly thank you for this great library :)

I would like to request an enhancement to add **STM_PORTC_DATA_BUS** please as it is far less contestable than Port A and Port B and the performance improvement when using a port rather than individual pins is dramatic.

**So why not use the ones provided you may ask:**

The answer is availability for use on a PCB; Port B has the boot pins that are best not hardwired anywhere and the SPI pins. PortA is more available but anyone using Deep sleep will need PA0 as the WKUP pin and may need the Serial on PA3/PA4.
PortC on the other hand is just digital and ADC pins and would be the most commonly used ones if it was an option. 

**User_Setup.h**
```
// STM32 8 bit parallel only:
// If STN32 Port A or B pins 0-7 are used for 8 bit parallel data bus bits 0-7
// then this will improve rendering performance by a factor of ~8x
//#define STM_PORTA_DATA_BUS
//#define STM_PORTB_DATA_BUS

//#define STM_PORTC_DATA_BUS
```

Thanks in advance for your help

toogooda, 2021-08-28:
Actually was just looking at the library to see how hard it might be to implement I found some code that suggests PortC might already be implemented and it may just be missing as an option in User_Setup.h.

```
#elif defined (STM_PORTC_DATA_BUS)
  #if defined (STM32F1xx)
    if (mode == OUTPUT) GPIOC->CRL = 0x33333333;
    else GPIOC->CRL = 0x88888888;
  #else
    if (mode == OUTPUT) GPIOC->MODER = (GPIOC->MODER & 0xFFFF0000) | 0x00005555;
    else GPIOC->MODER &= 0xFFFF0000;
  #endif

```
Do I just need to add #define STM_PORTC_DATA_BUS in the user setup to take advantage of it or is the code just a partial implementation?

Thanks

toogooda, 2021-09-09:
As per above is STM_PORTC_DATA_BUS supported already? if so I can close this and test it when the new hardware arrives.

Thanks for your help

Bodmer, 2021-09-14:
Yes, port C capability is in the code for STM32F processors and was added on request. I have not done much testing though but one user said it worked OK.

Raise a new issue if you have problems.

closed 2021-09-14 by Bodmer

toogooda, 2021-09-15:
Thank you I will do some testing :)