Bodmer/TFT_eSPI #1313: Problem with spr.printToSprite() cutting off character

fdrake97, 2021-08-29:
I am printing some numbers to a screen using `spr.printToSprite()`, and sometimes the last digit does not show up (see image below).

Below is the code snippet for this particular case. The time it should be printing is 35.018 and so `t` in this case is `"35.018"` and it has happened with several other numbers: 28.212, 30.517, 34.716, 35.018, 39.416. All numbers have a 1 in the hundredths place. I am almost 100% confident that `spr.printToSprite()` is receiving the correct string: you can tell by the image that the x coordinate of the tft cursor position given by `790-getTextWidth(" " + t,26)` is correct. If printToSprite had received just "35.01" instead of "35.018" then the number should be completely right-aligned.

Another thing I noticed is that little stray pixel on the right, which is what was "underneath" before the 35.01 was displayed. It almost seems like the rectangle containing the 35.018 is 1 pixel too small, resulting in the rightmost number being cut off.

```
drawTime(" " + t, 790-getTextWidth(" " + t,26), tft.getCursorY()+getTextHeight(30)+15, TFT_WHITE, TFT_BLACK, 26);

void drawTime(String timeString, int x, int y, uint16_t textColor, uint16_t bgColor, int fontSize){
  if (fontSize == 26){
    spr.loadFont(NotoSans26);
  }
  else if (fontSize == 30){
    spr.loadFont(NotoSans30);
  }
  spr.setTextColor(textColor, bgColor);
  tft.setCursor(x,y);
  spr.printToSprite(timeString);  // WHERE IT GOES WRONG (I think)
  spr.unloadFont();
}

// get width of string in given font size
int getTextWidth(String str, int fontSize){
  
  int w;
  if (fontSize == 26){
    spr.loadFont(NotoSans26);
    w = spr.textWidth(str);
    spr.unloadFont();
  }
  else if (fontSize == 30){
    spr.loadFont(NotoSans30);
    w = spr.textWidth(str);
    spr.unloadFont();
  }
  return w;
}

int getTextHeight(int fontSize){
  
  int h;
  if (fontSize == 26){
    spr.loadFont(NotoSans26);
    h = spr.fontHeight();
    spr.unloadFont();
  }
  else if (fontSize == 30){
    spr.loadFont(NotoSans30);
    h = spr.fontHeight();
    spr.unloadFont();
  }
  return h;
}
```

[image]

Bodmer, 2021-09-13:
I think this occurs in particular circumstances. Can you provide a complete sketch (as a zip file) that demonstrates the problem.

fdrake97, 2021-09-13:
Yes, I will send you a sketch when I get home later today. After further testing it seems to happen every time a 1 is the second to last character. This is with the Noto fonts (Noto Sans Semibold), in all font sizes that I've used. Haven't tried other fonts.

fdrake97, 2021-09-14:
[printToSprite_bug.zip](https://github.com/Bodmer/TFT_eSPI/files/7165470/printToSprite_bug.zip)

It happens with some characters and not others. For example "1A" didn't work but "1a" did.

Bodmer, 2021-09-14:
I can see what is happening, the library is not using the correct width for the sprite so some characters are wrapping to the next line in the sprite, but the sprite is not tall enough to see this has happened.

If wrapping is switched off then I can see the sprite width calcculation is 1 pixel too small with some characters. eg.

```
#include "NotoSans18.h"

#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite spr = TFT_eSprite(&tft);

void setup(){

  //need to turn on screen
  //pinMode(21,OUTPUT);
  //digitalWrite(21,HIGH);
  
  delay(500);
  tft.init();
  delay(200);
  tft.fillScreen(TFT_BLUE);
  delay(200);
  tft.setRotation(1);
  spr.loadFont(NotoSans18);
  spr.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setCursor(5,5);
  spr.setTextWrap(false);
  spr.printToSprite("1A"); //replace A with other characters. 
  spr.unloadFont();
  
}

void loop(){}
```

Addind a space at the end is one workaround in the short term until I have this fixed.

fdrake97, 2021-09-14:
Yes, this is what I've been doing, and it's not too bad. The only issue is when there is something immediately to the right of the rightmost character, especially with a larger font that has a decently wide space character.

Bodmer, 2021-09-15:
Try using spr.setTextWrap(false); in setup() as an alternative temporary workaround, that may give better results in your situation.

fdrake97, 2021-09-15:
Ok. If I'm understanding correctly that will draw everything correctly, just without the rightmost column of pixels. Shouldn't be too noticeable.

Bodmer, 2021-12-12:
This is fixed now at version 2.3.88.

commit a474f4cd (2021-12-12): Fix #1313
closed 2021-12-12 by Bodmer