Bodmer/TFT_eSPI #1318: rp2040 and ST7735 not able to get working

nruffilo, 2021-08-31:
I'm using a pi pico with the https://www.adafruit.com/product/2088 breakout board.  I can confirm the wiring works with the standard adafruit_gfx library graphicstest, and used the same configuration with TFT_eSPI and was unable to get any of the examples working.  

I tried all of the following:
#define ST7735_DRIVER 

// #define ST7735_INITB
//#define ST7735_GREENTAB
//#define ST7735_GREENTAB2
//#define ST7735_GREENTAB3
#define ST7735_GREENTAB128    // For 128 x 128 display

Ideally I'd love to get the animated eyes working - and your DMA rewrite (as opposed to the original zeroDMA implementation) is extremely exciting.  Any tips you might have would be greatly appreciated, or if you can let me know what code in your library may be the issue I can attempt to find/solve it myself and contribute back.  Thank you!

BTW - code compiles and uploads, but screen remains white.

djairjr, 2021-09-03:
@nruffilo , I suggest you make the following changes to the Setup47_ST7735.h file, which is inside the User_Setups folder. Or create a copy of this file, naming Setup_##_Rp2040_ST7735.h, for example. Then, make a reference for this file in User_Setup_Select.h, creating a line with #include <User_Setups/Setup_##_Rp2040_ST7735.h>

```
// Config for two ST7735 128 x 128 displays for Animated_Eyes example

#define ST7735_DRIVER     // Configure all registers

#define TFT_WIDTH  128
#define TFT_HEIGHT 128

// #define ST7735_INITB
// #define ST7735_GREENTAB
// #define ST7735_GREENTAB2
 #define ST7735_GREENTAB3
// #define ST7735_GREENTAB128    // For 128 x 128 display
// #define ST7735_GREENTAB160x80 // For 160 x 80 display (BGR, inverted, 26 offset)
// #define ST7735_REDTAB
//#define ST7735_BLACKTAB
// #define ST7735_REDTAB160x80   // For 160 x 80 display with 24 pixel offset

//#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

//#define TFT_INVERSION_ON
//#define TFT_INVERSION_OFF

// Changed to Raspberry Pi Pico Pinout
#define TFT_MISO 0 // RX PIN
#define TFT_SCLK 2
#define TFT_MOSI 3 // TX PIN
#define TFT_CS   20  // Chip select control pin
#define TFT_DC   18  // Data Command control pin
#define TFT_RST  19  // Reset pin (could connect to Arduino RESET pin)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
//#define SPI_FREQUENCY  40000000

#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

// #define SUPPORT_TRANSACTIONS
```
See if it works.

nruffilo, 2021-09-03:
This worked!  You are amazing.  Thank you so much!  I did have to swap to RGB and not BRG but that was an easy find/fix.

djairjr, 2021-09-04:
Perfect!

Did you change this line?
```
//#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue <- Uncomment this
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red <- Comment this
```
Good!

nruffilo, 2021-09-04:
Yes, the display I have is RGB

Bodmer, 2021-09-13:
@djairjr 
Thanks for helping resolve this.

closed 2021-09-13 by Bodmer